ros-industrial-attic/ur_modern_driver #237: lowbandwidth controller does not arrive (in time)

v4hn, 2018-11-09:
We tried @potiuk's (rather recently merged) lowbandwidth trajectory follower via `use_lowbandwidth_trajectory_follower:=true` together with a standard MoveIt pipeline generating trajectories.

In practice and on our ur5, this works quite well only about half the time.
For many, apparently especially shorter, trajectories, the action server does not succeed but keeps blocking, even when the arm perceivably stopped.
At some point (with the default settings quite soon) MoveIt aborts the `FollowJointTrajectory` action server because it "takes too long", i.e. the maximum expected duration for trajectory execution was exceeded.

We did not investigate further yet, but this could be related to 
https://github.com/Zagitta/ur_modern_driver/pull/9#issuecomment-357633072 where @axelschroth demonstrated that the implementation stops before reaching the goal state.

We don't see these problems with `use_lowbandwidth_trajectory_follower:=false`.

Any ideas or comments? Does it make sense to look into this further or will the upcoming ROS driver promised by UR supersede this implementation anyway? @gavanderhoorn

potiuk, 2018-11-10:
> Zagitta#9 (comment)

Unfortunately, I have no access to UR robot any more (changed jobs :( ) so
I can't help too much with testing any fixes :(. What I can recommend at
this time is to increase the tolerance of MoveIt - you can modify both
percent margin and additional time added to maximum time expected - it was
somewhere in moveit configuration.

I also recall that in our python implementation which we finally settled on
we found out that we need to actively read RTDE messages all the time,
otherwise there was a backlog of messages to read when you start
connecting. I don't there is a similar problem here as RTDE is not used.

J.

v4hn, 2018-11-12:
> changed jobs :(

Too bad! Thank you for your contributions. :)

> What I can recommend at this time is to increase the [time] tolerance of MoveIt

We did that, but the FJT server simply does not arrive in many cases and
the action does not return for *minutes* when the trajectory is only a few seconds long.
This is not usable in practice.

gavanderhoorn, 2019-11-23:
> Does it make sense to look into this further or will the upcoming ROS driver promised by UR supersede this implementation anyway?

in the end: yes.

But as noted in the deprecation announcement, the "low bandwidth" mode is not supported right now in `ur_robot_driver` (although it does have mitigations in place for handling dropped packets).

closed 2019-11-23 by gavanderhoorn

v4hn, 2019-11-23:
it might make sense to keep this issue open in archive, as a community contribution for this to the new driver will likely take this as a basis.

thank you for all your efforts here!