ros-industrial-attic/ur_modern_driver #243: move() is not blocking

FlyingBeginner, 2018-11-29:
Hi,

I am using the ur_modern_driver package in combination with moveit to plan the UR5 motion. I found the following problem:
case a) When I initialize the robot using 'roslaunch ur_modern_driver ur5_bringup.launch' the move()-method is blocking.
case b) When I initialize the robot using 'roslaunch ur_modern_driver ur5_ros_control.launch' the move()-method is NOT blocking.

Is there a way to make the move()-method blocking using the ros-control bringup (case b)?

Thank you so much!
Best regards
Jan

gavanderhoorn, 2018-12-05:
The `ros_control` side of the driver relies on `joint_trajectory_controller` for its Action server. That particular implementation supports *trajectory replacement* (or: blending). That's why by default you can submit as many trajectories as you want: the controller will try to blend them all together.

The regular code paths in this driver do not support trajectory blending.

So the "blocking" you describe is actually a side-effect.

> Is there a way to make the move()-method blocking using the ros-control bringup

No, not that I'm aware of.

You'll have to make sure to check for completion of the goal you submitted to the `FollowJointTrajectory` action server (which is something you probably should do regardless).

gavanderhoorn, 2018-12-05:
As this is not something we can change in this package I'm going to close this issue.

If you feel this is wrong @FlyingBeginner please re-open and clarify.

closed 2018-12-05 by gavanderhoorn