ros-industrial-attic/ur_modern_driver #244: ROS 2 Support

davetcoleman, 2018-12-05:
I'm surprised to be the first to bring up this topic, but let's start the discussion on what a UR driver for ROS2 would look like. I ask because of my own initiatives for a MoveIt2 that would require a compatible robot driver.

What a search turned up for me:
- Erle Robotics [seems to have a demo](https://www.youtube.com/watch?v=3F28y8DYiUk) but it appears they never released it (probably because they are developing their own robot hardware now).  @rkojcev
- At ROSCon 2017 a [ROS2 demo used the UR5](https://github.com/Kukanani/picky_robot/blob/ros2/nodes/ur5_pusher.py#L47), but it used a simple Python library called [urx](https://github.com/SintefManufacturing/python-urx) that I do not believe has nearly the same feature set as the ur_modern_driver. In particular, that demo just used ``movel`` and ``translate_tool``. @Kukanani

Is anyone else working on this? 

A related effort is a very early proof of concept of [ros_control2](https://github.com/ros-controls/ros2_control), though it is not a port.

Kukanani, 2018-12-05:
Thanks for starting the conversation. Just to clarify, for my application, I actually made a ros2 fork of urx: https://github.com/Kukanani/python-urx. It was ultimately a set of hacks to get (some, any) manipulator to work with ROS2. As you mentioned, I think that this package is both better supported and has more features than urx.

gavanderhoorn, 2018-12-05:
I'm not aware of any ROS 2 ports of this driver.

In any case: a meaningful port would depend on Actions being available, so that would be something to look into after the release of ROS2 CC.

rkojcev, 2018-12-10:
@davetcoleman we did an initial port of this UR driver to ROS 2 almost 2 years ago but in that time ROS 2 was still in alpha ( I believe) and it did not have all the nice functionalities such as actions that ROS 1 had. After that we neglected that project, and focused on releasing our own solution.

davetcoleman, 2018-12-12:
An update from the ROS Industrial Consortium meeting in Germany today: Universal is releasing official UR drivers! But for ROS 1, likely landing late Spring 2019. They plan on making ROS 2 drivers in the future, but they cited most of their customers were still on ROS 1 so that was their first goal.

gavanderhoorn, 2018-12-12:
To be more precise: as is always a good idea, the driver will be "thin", so most of the interface parts should be relatively easily portable to ROS 2. Once it makes sense.

planetceres, 2019-02-19:
This conversation is a few months old -- has any progress been made on ROS2 drivers?  What criteria still needs to be met in order for ROS2 drivers to "make sense"?

gavanderhoorn, 2019-02-21:
> This conversation is a few months old -- has any progress been made on ROS2 drivers? What criteria still needs to be met in order for ROS2 drivers to "make sense"?

re: "make sense": this was not a comment on a ROS 2 version of `ur_modern_driver`, but on the design and approach taken for the *new* driver.

As to ROS 2 support for `ur_modern_driver`: now that actions have finally (sort of) landed in Crystal, it starts to become viable to look into ROS 2 support.

atyshka, 2019-03-12:
@davetcoleman, now that spring 2019 is upon us, any updates on the official ROS 1 driver?

davetcoleman, 2019-03-13:
To be clear, I'm not a spokesperson for UR. I was only sharing news I heard at a ROS Industrial conference from a UR representative. I've no updates

Prier, 2019-05-04:
In case your interested I have recently published a new C++ interface that uses the RTDE interface of the UR for sending and receiving data. The interface can also be used with python through bindings. The interface is meant to be usable with various robot frameworks, which is why the receive and control interface relies only on STL datatypes. One can choose to convert to STL types or simply rewrite the control and receive interfaces to the desired datatypes to avoid any overhead. The plan is to make a more complete ROS1 / ROS2 interface based on this project. 

You can find it here:
https://gitlab.com/sdurobotics/ur_rtde

Documentation with examples and API resides at:
https://sdurobotics.gitlab.io/ur_rtde/

gavanderhoorn, 2019-05-05:
@Prier: thanks for making your work OSS.

The current effort for the UR-backed driver includes something similar, so perhaps seeing where the overlap could be used to benefit both would be interesting. That would be up to UR though.

Prier, 2019-05-05:
@gavanderhoorn Yes I have also been briefly in contact with a person at UR and it would be great if their can use some of it for the new ROS interface.

atyshka, 2019-05-06:
@Prier @gavanderhoorn Any specifics from UR as to what to expect from this driver vs the current implementations? Also tentative release schedule?

Prier, 2019-05-06:
@atyshka you can see the presentation they  gave here: https://static1.squarespace.com/static/51df34b1e4b08840dcfd2841/t/5c4090b470a6adde64a153c2/1547735256433/2018.12.12_Anders_Billesoe_Beck.pdf but gavanderhoorn might know more than me :)

gavanderhoorn, 2019-05-06:
All the publicly available information on the new driver may be found in the presentation (slides and video) that Anders Billesø Beck gave at the ROS-Industrial conference in Stuttgart in December '18.

@Prier linked the slides.

schornakj, 2019-06-17:
I think a straight port of the current driver to ROS2 would be very useful -- I'd immediately benefit from having it for my projects, and I'm sure others would too. It sounds like the official UR-backed driver won't support ROS2 for some time.

I took a look through this driver's ROS code and the amount of work doesn't seem hugely intractable, especially if the scope of work is initially limited to only the functionality brought up by `ur_common.launch`.  Most of the changes to compiled code would be in `ur_ros_wrapper.cpp`.
- Reformulating the FollowJointTrajectory action server to use ROS2 actions (much less of an obstacle now that Dashing is released).
- Changing the RosWrapper class so it extends the ROS2 base node class.
- Switching to ROS2-style parameter lookups.
- Translating the launch files to ROS2 .launch.py's.
- Tweaking the ur_modern_driver package's CMakeLists.txt and package.xml to be compatible with both ROS versions.
- Porting the ur_msgs package to be compatible with both ROS versions.

Is there anything critical or blocking that I've missed?

Porting `ur_hardware_interface.cpp` and the alternative control strategies is a whole other can of worms, which I wouldn't mind putting off for the time being. In particular it looks like it would require porting the ur_description package: since it's heavily reliant on xacro (which isn't officially supported in ROS2) I'm not sure about the best way to proceed there.

gavanderhoorn, 2019-06-17:
@ipa-hsd / @ipa-led

gavanderhoorn, 2019-06-18:
@ipa-hsd & @ipa-led: I thought you might have some input here, seeing as you've already worked on a `ur_modern_driver` port.

ipa-hsd, 2019-06-18:
Thanks @gavanderhoorn for pointing this out. We did a `ur_modern_driver` port to ROS2.0 for the [ROS2/DDS tech workshop](https://rosindustrial.org/events/2019/05/06/ros-industrial-eu-19-spring-workshop). We had some issues while trying to send trajectories from MoveIt! but works fine if we directly send a `FollowJointTrajectoryAction` goal from a python script. We need some testing to fix this.

Currently the status as per @schornakj's list is:

- [x] Reformulating the FollowJointTrajectory action server to use ROS2 actions (much less of an obstacle now that Dashing is released).
- [x] Changing the RosWrapper class so it extends the ROS2 base node class.
- [ ] Switching to ROS2-style parameter lookups.
- [ ] Translating the launch files to ROS2 .launch.py's. (*half way through*)
- [ ] Tweaking the ur_modern_driver package's CMakeLists.txt and package.xml to be compatible with both ROS versions. (*not sure about this*)
- [x] Porting the ur_msgs package to be compatible with both ROS versions.

Additionally,
@ipa-led: can you comment on the changes you made for the `Time` API?

Also, we need to fix some git branch issues as @gavanderhoorn noticed.

I also noticed `robot_state_publisher` has been ported for ROS2, but for the workshop we still used ROS(1) version. We did not touch `ur_description` part either as pointed out earlier.

schornakj, 2019-06-18:
@gavanderhoorn @ipa-hsd  I think that for a message package cross-compatibility makes sense, since `ros1_bridge` requires matching message definitions in both the ROS and ROS2 workspaces. Since having a fully cross-compatible package [as discussed in this Discourse thread](https://discourse.ros.org/t/ros2-transition-strategy/9154) isn't really a core functional requirement I'll amend that item to:
- Tweaking the ur_modern_driver package's CMakeLists.txt and package.xml to be compatible with ~~both ROS versions~~ ROS2.

@ipa-hsd I've used the ROS2 version of `robot_state_publisher` and it works just like I'd expect. As a workaround for the ROS2 xacro parsing issue I just used `rosrun xacro xacro` to convert to URDF before running my system.

It sounds like there's already significant progress, which is very cool! I'd be interested in contributing if it makes sense.

darkdragon-001, 2019-06-22:
You could try the ROS2 `xacro` pull request: https://github.com/ros/xacro/pull/203

schornakj, 2019-07-10:
@ipa-hsd Any further thoughts or progress on the ROS2 migration? @gavanderhoorn and I talked about the state of things for a little bit during the ROS-I developer call this morning.

ipa-hsd, 2019-07-10:
Sorry about the delay guys. I will get this done this week.

kazuki0824, 2019-07-14:
I’m working on porting “universal_robot” into ROS2 Dashing here: https://github.com/kazuki0824/universal_robot/tree/dashing-devel

schornakj, 2019-08-15:
@ipa-hsd How is this going? Could you send me an email when you get a chance? I have some offline questions.

gavanderhoorn, 2019-11-23:
We will not be adding ROS 2 support here in `ur_modern_driver`.

Any efforts towards ROS 2 support should be focused on `ur_robot_driver` instead.

There are a few aspects of `ur_robot_driver` that could facilitate a ROS 2 port, and community members interested in this are strongly encouraged to *first* open an issue on the `ur_robot_driver` issue tracker to discuss potential approaches.

closed 2019-11-23 by gavanderhoorn