ros-industrial-attic/ur_modern_driver #245: Officially release kinetic refactor

gavanderhoorn, 2018-12-13:
This is a meta-issue to track and coordinate the release of the `kinetic-devel` refactor.

Tasks to be completed before that can be done:

 - ROS wiki:
   - [ ] short installation tutorial
   - [ ] overview of compatibility with controller and software versions / when to use which driver
 - repository (`readme`):
   - [ ] clarify status of different branches:
     - use `kinetic-devel` by default (it also builds on Indigo)
     - use `indigo-devel` (current `master`) if the old version is preferred
   - [ ] short introduction
   - [ ] status of the repository
   - [ ] where to go for help (issue tracker, ROS Answers, etc)
   - [ ] notes about performance
   - [ ] mention important compatibility information (possibly duplicate the ROS wiki section(s))
   - [ ] link to other related documentation (use with MoveIt, etc)
   - [ ] split out descriptions of the default, `ros_control` and low bandwidth traj follower variants
   - [ ] document how to build from source
 - [ ] Bloom a release (Kinetic, maybe Indigo (but release from `kinetic-devel`)

gavanderhoorn, 2018-12-13:
@miguelprada @ipa-nhg @ipa-mdl FYI.

ipa-nhg, 2018-12-17:
Some updates done:
- http://wiki.ros.org/ur_modern_driver
- http://wiki.ros.org/universal_robot (minor changes for consistency)
- https://github.com/ros-industrial/universal_robot/pull/393 (minor changes for consistency)

For the readme of this repository I suggest to move the **improvements**  section to a new markdown file or, at least, to the end of the readme. 

Also everything related to API, parameters and configuration usually is part of the ROS wiki and  not the readme of the repo. @gavanderhoorn what do you think? 

Basically for the readme we can take other ros-i repo readme as template, what about [fanuc](https://github.com/ros-industrial/fanuc/blob/indigo-devel/readme.md)?

gavanderhoorn, 2018-12-17:
> Some updates done:
> 
>     * http://wiki.ros.org/ur_modern_driver
>     * http://wiki.ros.org/universal_robot (minor changes for consistency)
>     * [ros-industrial/universal_robot#393](https://github.com/ros-industrial/universal_robot/pull/393) (minor changes for consistency)

great. Thanks.

> For the readme of this repository I suggest to move the **improvements** section to a new markdown file or, at least, to the end of the readme.

yes, perhaps we can even create a `docs` directory.

> Also everything related to API, parameters and configuration usually is part of the ROS wiki and not the readme of the repo. @gavanderhoorn what do you think?

I'm not against having it here in the readme, although I do agree information like that is typically part of the wiki pages for a package. I'm not sure, what do you think @miguelprada?

Advantage of having that information here on GH is that we keep everything in one place. We might even consider converting some of the readme content into proper documentation (using sphinx or something similar). Then refer to it from the wiki, similar to other packages (like MoveIt fi).

> Basically for the readme we can take other ros-i repo readme as template, what about [fanuc](https://github.com/ros-industrial/fanuc/blob/indigo-devel/readme.md)?

I'm ok with keeping the readme simple and similar to others. Separating-out the extensive sections about parameters, topics and additional functionality would already help significantly.

ipa-nhg, 2019-04-04:
@miguelprada @gavanderhoorn I would like to move this forward, concretely update documentation and release asap the repos.

What about docu?  
- Minimal README: versions supported, license badge and how to install (from release and from src) and start. (and if required warnings and limitations)
- Move the API to the ros wiki
- For the rest of already available documentation add a docu folder and move everything there including Zagitta thesis https://github.com/ros-industrial/ur_modern_driver/issues/268

opinions?

miguelprada, 2019-04-04:
Sounds good. It's going to be a challenge to make it really minimal, considering there's three different "modes" of execution for this driver, though :sweat_smile:

Are you willing to start with this? You need help with anything in particular?

miguelprada, 2019-04-04:
This would also require a new `universal_robot` release, right?

gavanderhoorn, 2019-04-05:
>     * Minimal README: versions supported, license badge and how to install (from release and from src) and start. (and if required warnings and limitations)

If with minimal you mean: a readme similar to other ros-industrial repositories without all the information on configuration and compatible system etc, then that makes sense to me.

>     * Move the API to the ros wiki

With ROS 2 in mind (and the changes to how documentation is going to be hosted and viewed by users): may I suggest to actually keep that information here in the repository, but just break up the current readme into a few files, one per 'subject'?

That would accomplish the same thing, we can link to it from the ROS wiki, but would make it easier to migrate that text to whatever ends up being the platform of choice for ROS 2 (which at the moment is sphinx + rst).

>     * For the rest [..] add a docu folder and move everything there including Zagitta thesis #268

I changed my mind about that: the thesis will be hosted by DTU, we can link to it. It's more than 1.5MB, isn't going to change and a big binary blob, so I would rather not host it in this git repository.

gavanderhoorn, 2019-04-05:
@miguelprada wrote:
> This would also require a new `universal_robot` release, right?

yes. I'll do that soon.

But I'd like to get #281 in first.

gavanderhoorn, 2019-11-23:
This package will therefore not be released through the ROS buildfarm.

closed 2019-11-23 by gavanderhoorn