ros-industrial-attic/ur_modern_driver #248: fail to  control real   UR5

nashdingsheng, 2018-12-29:
hi， i can not move the real UR5 robot , i believe i have connected to the robot and can derive the joint_stataes timely by topic /joint_states, the main working environment:
 UR5,firmware 3.4.5-100.,(kinetic-devel  ur_modern_driver) as the driver for hardware 
Winows 10 , i install ubuntu 16.04 on VMMWARE as a virtual machine to control, ROS-kinetic version.
after i git and catkin_make  the related packages（-b kinetic-devel ur_modern_driver, -b kinetic-devel universal_robot）,then 
i do :
1.roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.22（my ur5 robot ip）
2../test_move.py  （brought from  kinetic-devel branch）   
the robot does not move as the test_move.py intends to move. 
and i do as the readme suggested :
1.roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.22（my ur5 robot ip）
2.roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch
3.roslaunch ur5_moveit_config moveit_rviz.launch config:=true
 RIVZ can timely show the change of the robots states that i use VNC tool to login in  the robot to control, and i drag the coordinate system to a new place in RVIZ and  press "plan and execute", the robot does not move as well.
 I do not know what happened for this phenomenon ,and i also intend to repair as the suggestion for the similar problem suggested，could somebody please give me some advice,thank you.

gavanderhoorn, 2018-12-29:
I'm not saying this cannot work, but I see the following in your description:

 - Windows
 - Ubuntu in VMware
 - VNC

Those are all complicating factors, and don't help debugging.

Right now I would guess your VMWare is using a NATted virtual network which only allows outgoing connections by default. The driver needs to create a connection from the robot to the driver node in order to receive trajectory points. If that connection doesn't work, the robot won't move, even though `JointState` messages are being published.

First thing to try: configure VMWare to use a *bridged* network connection.

gavanderhoorn, 2019-01-02:
As this is (most likely) not an issue with the package in this repository I'm closing this issue.

Feel free to keep commenting on it though @nashdingsheng.

closed 2019-01-02 by gavanderhoorn

nashdingsheng, 2019-01-07:
gavanderhoon, thank you very much, today i attempt to change the network configuration to bridge network and ur5 is in control now. i was not in office  but at home welcoming my first new baby over the past one week and reply you until today,thank you very much for your help.

gavanderhoorn, 2019-01-07:
No problem.

Good to hear you got it to work.

As an alternative to bridged networking, you could also configure the NAT NIC to forward `tcp://:50001` to the IP of your ROS VM.