ros-industrial-attic/ur_modern_driver #249: enforce limits in ros_control hw interface

gavanderhoorn, 2019-01-04:
The [current implementation](https://github.com/ros-industrial/ur_modern_driver/blob/e2d81cf2b88237701931e02dbf05780d135c81c4/src/ros/hardware_interface.cpp) of the `hardware_interface` for `ros_control` does not appear to be checking nor enforcing joint limits.

This should be added.

PS: labelled as an *enhancement*, but perhaps this should really be considered a *bug*.

gavanderhoorn, 2019-01-04:
From: https://github.com/ros-controls/ros_controllers/pull/393#issuecomment-448121951.

davetcoleman, 2019-01-17:
Thanks for reporting @gavanderhoorn 

Note we do have a [joint_limit_interface](https://github.com/ros-controls/ros_control/tree/melodic-devel/joint_limits_interface) in ros_control, it just needs to be integrated into the ur_modern_driver. Here's some [example code](https://github.com/PickNikRobotics/ros_control_boilerplate/blob/melodic-devel/src/generic_hw_interface.cpp#L121)