ros-industrial-attic/ur_modern_driver #250: sensor_msg/Joint_states values without precision

VascoLopes, 2019-01-04:
Hello,

I've been subscribing to `/joint_states` to receive values regarding velocity, effort, and position while a real UR3 performs a pick and place task repeatedly.

Example of values with imprecision:
```
header: 
  seq: 28119
  stamp: 
    secs: 1546536761
    nsecs: 942488273
  frame_id: ''
name: ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
position: [1.0115061724275923e-25, 5.33615435e-315, 5.32831366e-315, 1.491668503445072e-154, 1.4916685014862012e-154, 1.4916685034458053e-154]
velocity: [2.000000478933958, -2.681562228132196e+154, 5.3283246e-315, 2.000000478931979, -5.3283369e-315, -1.491668507438338e-154]
effort: [5.304989477e-315, 1.5871121005e-314, 6.6794773466055355e+69, 9.545934740755065e-207, -9.476754267470889e+251, -3.62914187213457e+229]
```

What can be causing this problem? I'm running the robot from a laptop with:
rosrun ur_modern_driver ur3_bringup.launch robot_ip:=10.0.1.2

and capturing the values directly from the topic to a file.

Thank you.

gavanderhoorn, 2019-01-06:
I'm not sure what you mean by "imprecision", but the `position`, `velocity` and `effort` fields seem to contain strange values.

You're going to have to provide more information here:

 - which version of the code are you running? Current `master`, or `kinetic-devel`?
 - what version is the system software of the controller?

Also: are you running this in a VM?

VascoLopes, 2019-01-06:
Hi,

Sorry for my weak description.
I'm currently using `master`
The system software is the latest release and I'm running it on ubuntu 14.04 without a VM

VascoLopes, 2019-01-07:
Changing to kinetic-devel solved the problem (at least, until now)

For completeness:
I have a UR3/CB3 with a RG2
URSoftware : 3.8.0.61336

Thank you again

closed 2019-01-07 by gavanderhoorn