ros-industrial-attic/ur_modern_driver #251: roslaunch ur_modern_driver STOP, but NO Error

songWell, 2019-01-06:
Hi!
When I connected  UR5, ping was smooth, but roslaunch stopped, and no error was reported.
The following is my terminal information。
<well@well:~$ ping 192.168.1.101
PING 192.168.1.101 (192.168.1.101) 56(84) bytes of data.
64 bytes from 192.168.1.101: icmp_seq=1 ttl=64 time=0.209 ms
64 bytes from 192.168.1.101: icmp_seq=2 ttl=64 time=0.125 ms
64 bytes from 192.168.1.101: icmp_seq=3 ttl=64 time=0.125 ms
64 bytes from 192.168.1.101: icmp_seq=4 ttl=64 time=0.121 ms
64 bytes from 192.168.1.101: icmp_seq=5 ttl=64 time=0.127 ms
64 bytes from 192.168.1.101: icmp_seq=6 ttl=64 time=0.152 ms
^C
--- 192.168.1.101 ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 4997ms
rtt min/avg/max/mdev = 0.121/0.143/0.209/0.031 ms>

<well@well:~$ roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.101
... logging to /home/well/.ros/log/f5c9bbf4-11b1-11e9-b1a5-b888e3f22812/roslaunch-well-20474.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: pi
when processing file: /home/well/ur_ws/src/universal_robot/ur_description/urdf/ur5.urdf.xacro
included from: /home/well/ur_ws/src/universal_robot/ur_description/urdf/ur5_robot.urdf.xacro
started roslaunch server http://well:39748/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /ur_driver/base_frame: base
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/robot_ip_address: 192.168.1.101
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/tool_frame: tool0_controller

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [20489]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f5c9bbf4-11b1-11e9-b1a5-b888e3f22812
process[rosout-1]: started with pid [20502]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [20513]
process[ur_driver-3]: started with pid [20516]>

roslaunch just stopped here. I really don't know what the problem is, it just worked three days ago. Can someone help me? Thank you very much!!!

songWell, 2019-01-06:
The format of the information is a bit problematic， sorry.But I can not edit it

gavanderhoorn, 2019-01-06:
Please show us the output of `env | grep -i ros | sort` and `ifconfig`.

songWell, 2019-01-06:
Thank you for your help.
It seems ok.
When I launched it last time, it had more [INFO] in the terminal. But this time, it doesn't. So I thought there must be something wrong. However, when I run test_move.py, UR5 moved. 
 It seems ok, though I don't know what problem is.
Thanks again.

closed 2019-01-06 by songWell