ros-industrial-attic/ur_modern_driver #253: Goal error: Received a goal with an invalid number of velocities

lakshmip001, 2019-01-11:
I have a joint trajectory obtained from trajopt motion planning package but when publish the trajectory on UR5 robot there is no motion and results in the above error . But when I use only rviz I can see the motion of the robot. Kindly please guide me how I can solve this error.

gavanderhoorn, 2019-01-11:
Assuming you are using `kinetic-devel`, the incoming trajectory is failing this test:

https://github.com/ros-industrial/ur_modern_driver/blob/e2d81cf2b88237701931e02dbf05780d135c81c4/src/ros/action_server.cpp#L169-L174

There's not much we can do about that here, in this package. You'll have to make sure to not submit goals with ill-formed trajectories.

gavanderhoorn, 2019-01-11:
As this does not appear to be an issue with this package I'm going to close this, but please feel free to keep commenting on it.

closed 2019-01-11 by gavanderhoorn