ros-industrial-attic/ur_modern_driver #257: Premature success using standard trajectory follower and no deacceleration in final moments using lowbandwidth_trajectory_follower.

janisa9, 2019-01-23:
- **ROS Kinetic, Ubuntu 16.04**
- **Universal Robots Software 3.5.1.10661**
- **UR5**

**1. Issue**

While using ur_modern_driver standard trajectory follower together with MoveIt, I encountered that 

`moveit::planning_interface::MoveItErrorCode::SUCCESS`

from

```
move_group_ptr->plan(my_plan);
success = move_group_ptr->execute(my_plan) == moveit::planning_interface::MoveItErrorCode::SUCCESS;
```

gives premature success that results in **C204A3: Protective Stop: Invalid setpoints: Sudden stop**. In result, there is a rapid stop, but if the protective stop isn't triggered after this stop the given trajectory stops really hard and next one starts as can be seen in this picture ->

**Image 1**
[image]

although it should look something like this (letting trajectory finish with delay insertion after recievieng success from moveit) ->

**Image 2**
[image]

Basically it should work as follows -> Trajectory execution -> when trajectory ends I get success and start to execute next trajectory. But happens this -> Trajectory execution -> Premature success of this trejectory -> next trajectory execution is started before previous one is ended that results in sudden/rapid stop. I don't know if it is connected with **ur_modern_driver** or **MoveIt**, so I am confused where I should be looking for a solution, because **lowbandwidth_trajectory_follower** doesn't give me premature success what leads me to **2. Issue**. It know it is somehow cnnected with that MoveIt trajectory execution time is lower than actual trajectory execution time, but I don't where and how it happens. Is it issue or I am just missing something? 

**2.Issue**

Second issue happens when **lowbandwidth_trajectory_follower** is used, the same movement code as with standard trajectory follower is used. As can be seen in image 3 there is no deaccelaration in final moments of trajectory what results in stop that is not soft as it should be - Image 2. 

**Image 3**
[image]

Again is this issue or I am again just missing something?

If any additional information is needed, please don't hesitate to ask, I would really apreciate any hints and information about this situation.

janisa9, 2019-01-25:
Any ideas?

janisa9, 2019-02-01:
I am not using `ros_control`mode.

Is test_move.py good enough to test driver behaviour without MoveIt!? With test_move.py I moved the robot through same positions as with MoveIt! and got almost the same picture - that in final moments there are really high deaccelaration (not no deacceleration, as I mistakenly wrote in original question), the only difference is that now I used a simulated robot, but same UR robot software (will try on real one later, if it changes anything). 

[image]

Is this ok?

gavanderhoorn, 2019-02-26:
Apologies for the delayed response, but unfortunately I don't know what is the cause of this right now.

I also don't have access to a CB3 with Polyscope 3.5.1, so I cannot reproduce.

If you're not using `ros_control` mode, then it would seem there could be something wrong in the sampling / trajectory generation.

I would also try to make sure the robot can actually complete the motion you're asking of it in 2 seconds.

gavanderhoorn, 2019-11-23:
If this is also still a problem with `ur_robot_driver`, please open a new issue on its issue tracker.

closed 2019-11-23 by gavanderhoorn