ros-industrial-attic/ur_modern_driver #258: Latency(?) issues working with UR10/CB3

miguelprada, 2019-01-25:
I've been having some latency(?) issues with a pair of UR10s controlled by respective CB3 controllers.

The observable symptoms when using the position interface, are the kind of catch-up motions that lead to implementing the low bandwidth control mode. When using the velocity interface the system starts to be very jerky and sometimes even unstable, depending on the specific controller being used.

I'm using the very same setup to test these robots and other UR10s controlled by CB3.1:
- PolyScope version 3.5
- Laptop running `kinetic-devel` version of the driver
- Direct wired connection from the laptop to the CB boxes
- Using `joint_trajectory_controller` to perform some simple motions

The only differences between the "faulty" robots and the ones that work well are the firmware versions of the Safety Control Board, and the controllers being CB3 vs CB3.1.

I already discarded the SCB being the issue by switching these boards between the robots, and the issues still show up.

One extra thing I noticed is that the problem gets worse the longer the robots are on. Whenever the performance degrades, rebooting the URs makes them behave well for a while.

I'd like to try connecting the "faulty" robots to the CB3.1 controllers, but I unfortunately don't have access to those anymore, so this will have to wait.

Another test I still didn't make is to try the low bandwidth mode, but this is not a permanent solution for me, since some of our use-cases require closed-loop control using FT sensors. These, by the way, work fine with the CB3.1-running UR10s (and another UR5 with CB3.1, for that matter).

I wanted to bring this up in case someone else having similar issues (or not) can think of additional tests I could run to try to diagnose and hopefully fix the problem.

gavanderhoorn, 2019-01-25:
Hm. That doesn't sound too nice.

I assume you have tried it, but did altering the `servo_j_time` change anything in the observed behaviour?

miguelprada, 2019-01-25:
> I assume you have tried it, but did altering the servo_j_time change anything in the observed behaviour?

Actually, I haven't. I tried playing with the time parameter in `speedj`, but not with the `servoj` one. Adjusting that value to a 0.02 as opposed to the 0.008 [used by default for version 3.5](https://github.com/ros-industrial/ur_modern_driver/blob/e2d81cf2b88237701931e02dbf05780d135c81c4/src/ur/commander.cpp#L149) did improve performance a bit, but still not on par with the other robots. This, in any case, increased my interest in #183.

Also, this was what initially lead me to investigate the possibility of the SCB firmware version being the determining factor to choose the right value for the speedj time parameter, but that ended up leading me nowhere.

I'll try to experiment a bit with `servo_j_time` when I have access to the robots, since they are quite busy these days.

miguelprada, 2019-02-01:
Just a quick update here. Yesterday we were able to borrow a CB3.1 and test it with our robots and they worked flawlessly. It seems that the issue comes from the CBs, then. Whether its a general problem with CB3 models or just with certain specific controllers remains to be seen (after all both our faulty CBs were bought together and may belong to a given faulty batch).

We're going to upgrade the CB motherboards to 3.1 and see if that fixes the issues for us.

gavanderhoorn, 2019-02-01:
Thanks for the update.

Have you had any contact with UR about this?

miguelprada, 2019-02-01:
> Have you had any contact with UR about this?

I got in touch, yes, but still haven't heard from them.

miguelprada, 2019-02-19:
Quick update on this one. Upgrading our CB3 boxes to CB3.1 got rid of the problem. I´m still not sure of what was going wrong, whether its a general issue with CB3 or whether there was a problem with our particular instances being faulty.

In any case, it is very unlikely I will continue to dig into this, so I will close the issue for now.

closed 2019-02-19 by miguelprada