ros-industrial-attic/ur_modern_driver #259: ros_control vel_based_pos_traj_controller leads to extemely slow UR5e motion, sometimes with error C204A2 protective stop.

DavidYaonanZhu, 2019-01-30:
Thanks for every one for implementing this wonderful driver.
The ordinary driver and **ros_control pos_based_pos_traj_controller** works really great for my teleoperation applycation (**UR5e**).  Now I want to implement impedance and admittance control for the robot that need to send velocity command.
However, when I try  to use **ros_control vel_based_pos_traj_controller** with Moveit and rqt_joint_trajectory_controller, the robot gives error  **C204A2** protective stop: path sanity check failed: inconsistency between target position and speed.
Then, I decreased the speed to 90% in teach pendant, the robot moves extremely slow. Is this the problem of PID gains?

miguelprada, 2019-01-30:
> However, when I try to use ros_control vel_based_pos_traj_controller with Moveit and rqt_joint_trajectory_controller, the robot gives error C204A2 protective stop: path sanity check failed: inconsistency between target position and speed.

That's an error that I haven't seen before. Which version of PolyScope are you running? A quick google search shows a bug that may trigger that error being [(partially) fixed in 3.2](https://www.universal-robots.com/how-tos-and-faqs/faq/ur-faq/release-note-software-version-32xxxxx/).

> Then, I decreased the speed to 90% in teach pendant, the robot moves extremely slow. Is this the problem of PID gains?

You should leave the speed slider in the pendant to 100%. Otherwise, the velocity commands will be dialed down inside the controller, causing all sorts of trouble (e.g. you'll constantly get path tolerance violations when using `joint_trajectory_controller`).

DavidYaonanZhu, 2019-01-31:
@miguelprada Thank you very much for your reply. I am using Polyscope 5.1.0 which is the control box for E series. 
1) Actually, at first I am getting the error on URSim simulator, which is also version 5.1.0.  Then I scrolled down the speed slider to 90%, the error disappear though robot moves extremely slow.
2) Then, I use real robot for testing the vel_based controller, speed slider is 100%, and I use rqt_joint_trajectory_controller sending command to only one joint (wrist3). 
The movie is here: a) https://youtu.be/oOjP526C_14, b)https://youtu.be/Y1anhWs4AZY
In a) the robot is moving extremely slow with "zzz" sound and overshoots my goal angle and several seconds later finally a protective stop.
In b) I am enabling the robot again and again, it moves step by step.
3) I also noticed that the PolyScope  keep flashes "Running" and "Stopped".
Errors I encountered are, C204A2 and C204A3.

Could it be something wrong with PID value?

miguelprada, 2019-01-31:
Sorry, I went over your initial comment too fast and didn't notice you were using a UR5e.

First things first, I guess you're using the branch in #216, right?

In the very preliminary tests I ran on the same robot, I also faced issues with the velocity interface. My best guess is what I wrote in [this comment](https://github.com/ros-industrial/ur_modern_driver/pull/216#issuecomment-453175617). If I'm right, you won't be able to use that mode without some serious refactoring of the driver code.

I would be very happy to be proven wrong, though. :wink:

DavidYaonanZhu, 2019-02-01:
Will this speedj problem regarding to UR E-series going to be fixed in future?

gavanderhoorn, 2019-02-01:
@DavidYaonanZhu wrote:
> Will this speedj problem regarding to UR E-series going to be fixed in future?

Most likely not without community contributions.

The new driver -- announced by UR [here](https://www.youtube.com/watch?v=gjrnOdymUUE) -- will work with CB3 and e-series and will support all of this natively.

DavidYaonanZhu, 2019-02-01:
@gavanderhoorn That sounds wonderful, I watched the video and sent them an email for beta.

gavanderhoorn, 2019-02-03:
The beta has not been announced yet, as the project is still underway.

Please be patient.

UR will contact you when it's time.

DavidYaonanZhu, 2019-03-01:
@miguelprada @gavanderhoorn 
Actually, I hope the velocity control can be reimplemented for E-series robots.
I want to make contribution to this issue, but I need to understand the structure of this driver first.

gavanderhoorn, 2019-03-01:
Please see my comment on #267.

Also please be aware that your work would parallel what is currently being done by UR and FZI. I'm not saying you shouldn't do anything, just wanted to draw your attention to that.

As to velocity control for e-series: I have a hunch that making sure `ur_modern_driver` gets real-time priority (setting scheduler params should be added to the node), you use a real-time capable kernel and the parameters dealing with `speedj` time are tuned properly should get us quite far.

GDUTwd, 2019-06-17:
> Thanks for every one for implementing this wonderful driver.
> The ordinary driver and **ros_control pos_based_pos_traj_controller** works really great for my teleoperation applycation (**UR5e**). Now [...]

sir, i want to know how you use ros_control pos_based_pos_traj_controller,i have read many data,but i dont know how tu use it to control ma robot.it will be great if you can give me some advise.

gavanderhoorn, 2019-11-23:
For all CB3 and e-series robots: please use the new `ur_robot_driver` instead of `ur_modern_driver`.

Refer to the announcement linked in my previous comment for more information.

closed 2019-11-23 by gavanderhoorn