ros-industrial-attic/ur_modern_driver #260: roslaunch ur_modern_driver ur5_bringup.launch  CONNECTION SETUP FAILED

Anastasiamor100, 2019-02-06:
PLEASE HELP!
I am trying to launch:
 ```roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.102 [reverse_port:=REVERSE_PORT]```

However am getting the following error: 
```
[ERROR] [1549412928.839559142]: Connection setup failed for 192.168.1.102:30003
[ INFO] [1549412928.839573683]: Setting up connection: 192.168.1.102:30001
[ERROR] [1549412928.839593882]: Connection setup failed for 192.168.1.102:30001
[ WARN] [1549412928.839602715]: Failed to read from stream, reconnecting in 1 seconds...
```
I am able to ping the robot obtaining this:
```
PING 192.168.1.102 (192.168.1.102) 56(84) bytes of data.
64 bytes from 192.168.1.102: icmp_seq=1 ttl=64 time=0.019 ms
64 bytes from 192.168.1.102: icmp_seq=2 ttl=64 time=0.032 ms
```
I have set up following this tutorial http://wiki.ros.org/universal_robot/Tutorials/Getting%20Started%20with%20a%20Universal%20Robot%20and%20ROS-Industrial 

How do I fix this issue?

Anastasiamor100, 2019-02-06:
Following from this I have used a static network and used the IP address from the setup (3.2 of the tutorial) as the IP address of the robot. Is this correct?

gavanderhoorn, 2019-02-06:
@Anastasiamor100 wrote:
> PLEASE HELP!

please don't do this. We already know you are looking for help, for otherwise you would not have posted here. Adding such lines to issues doesn't motivate anyone to help you any sooner.

Second: please provide more information on your setup. Right now we don't know anything, apart from that you're trying to start the launch file for a UR5.

So tell us:

 - which OS you are using
 - which version of ROS you are using
 - are you running things in a virtual machine (VMWare, VirtualBox, Parallels)?
 - which version of `ur_modern_driver` you are using (`master` branch? `kinetic-devel`?)
 - which controller your robot has (CB2, CB3, e-series)
 - which version of the system software/Polyscope it is running (`1.8.x`, `3.x` or a `5.x` variant)

> Following from this I have used a static network and used the IP address from the setup (3.2 of the tutorial) as the IP address of the robot. Is this correct?

Only if the host you are using to run `ur_modern_driver` is also part of the `192.168.1.x` network.

> I am trying to launch:
 ```roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.102 [reverse_port:=REVERSE_PORT]```

Are you literally typing `[reverse_port:=REVERSE_PORT]`?
If you are, just don't specify the `reverse_port` argument, ie start the launch file with just:

```
roslaunch --screen ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.102
```

If that doesn't work, copy-paste *all* output into a comment here and we can try to resolve your problem.

gavanderhoorn, 2019-02-06:
And I'm going to assume that this is not an issue with the pkg in this repository, but a user configuration one, so I'll close this ticket.

That doesn't mean we can't solve your configuration issue, so please feel free to keep commenting on the issue.

closed 2019-02-06 by gavanderhoorn