ros-industrial-attic/ur_modern_driver #261: error Action client not connected: /follow_joint_trajectory in UR10

marcoojer, 2019-02-07:
Hi I'm trying to move the UR10 via ros_control and I get the following error

`[ WARN] [1549552851.461456877]: Waiting for /follow_joint_trajectory to come up
[ INFO] [1549552854.918036995]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-kinetic-moveit-ros-planning-0.9.15/planning_scene_monitor/src/planning_scene_monitor.cpp:498
[ INFO] [1549552855.500717467]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ WARN] [1549552857.461723596]: Waiting for /follow_joint_trajectory to come up
[ERROR] [1549552863.462087537]: Action client not connected: /follow_joint_trajectory
`

I'm launching the following files:
`roslaunch ur_modern_driver ur10_ros_control.launch  robot_ip:=[MY IP]
roslaunch ur10_moveit_config ur10_moveit_planning_execution.launch`

When I used `ur10_bringup` it is connecting fine, and I can move the robot. However, with the  `ur10_ros_control` not. it seems like the node `move_group` is not publishing the \follow_joint-trajectory topic.

Anyone has any idea about how to solve it? Thanks in advance

janisa9, 2019-02-11:
Hi!

Have you followed read.me and done the following?

To use ros_control together with MoveIt, be sure to add the desired controller to the controllers.yaml in the urXX_moveit_config/config folder. Add the following

```
controller_list:
 - name: /vel_based_pos_traj_controller #or /pos_based_pos_traj_controller
   action_ns: follow_joint_trajectory
   type: FollowJointTrajectory
   default: true
   joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
```

Correct me if I am wrong but i think this could be your issue.

marcoojer, 2019-02-14:
It was exactly that. Thank you very much!!

closed 2019-02-14 by marcoojer

rijalmadhav12, 2022-03-24:
this is still not solving my issue..what could be the other possible things that need to be solved?

gavanderhoorn, 2022-03-24:
@rijalmadhav12: please first tell us which controller you have. A CB3 or an e-Series controller?

rijalmadhav12, 2022-03-28:
@gavanderhoorn I have CB3 controller(ur5). While using rviz  and moveit to plan real robot, it always says controller not found even after modifying controller.yaml

gavanderhoorn, 2022-03-29:
Could you please confirm you've seen [the deprecation notice in the readme](https://github.com/ros-industrial/ur_modern_driver#deprecation-notice)?

For anything which is not a CB2, you should not be using this driver.

Please use [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) instead.