ros-industrial-attic/ur_modern_driver #264: tool0_controller with ros_control doesn't get published

randomepicninja, 2019-02-26:
Hi! I'm using a UR3 and I need to know the positions accurately so I'd like to use the tool0_controller
I'm still relatively new to ROS so I apologize in advance if I only know the basics :|

I get to see the tool0_controller when I use the _ur3_bringup.launch_ but if I use the _ur3_ros_control.launch_, tool0_controller doesn't seem to get published. I don't see it when I check the rqt_tf_tree either.

**using bringup.launch**
[image]

**using ros_control**
[image]

I get the error message when I launch _ur3_moveit_planning_execution.launch_ 
```
ros.moveit_ros_planning.planning_scene_monitor: The complete state of the robot is not yet known.  Missing base_tool0_controller_floating_joint
```

I also tried looking further into the launch files and tried editing the _ur_common.launch_ and set the ``use_ros_control`` to "true" but it ended up with a worse tf tree

Is it that the using tool0_controller isn't supported when ros_control is used? or is it that I lack certain settings? Any help would be appreciated. Thanks :D

Oh, also, I'm using the one from the kinetic-devel branch

randomepicninja, 2019-02-26:
I just newly saw a pull request that addresses this issue #262 
I tried changing my rt_publisher.cpp and it worked :D thank you

closed 2019-02-26 by randomepicninja

miguelprada, 2019-02-26:
Related to #262.

rrapi, 2022-06-16:
Hi there!! I am using ur_robot_driver instead and I have the same issue. Also I can see the tool0_controller frame in RViz, but when I want to use that as reference frame i get this message:
`Link 'tool0_controller' not found in model 'ur5e_robot'`
Could you please help me to fix this?
@gavanderhoorn I saw the commit about this issue in #262 but the driver I am using is not the same

gavanderhoorn, 2022-06-16:
Please ask this on the `UniversalRobots/Universal_Robots_ROS_Driver` issue tracker instead.

While the drivers all share a development history, it's sufficiently different already for answers / information here not being relevant any more.