ros-industrial-attic/ur_modern_driver #266: CI checks fail for unreleased universal_robot features

miguelprada, 2019-02-28:
Some PRs here are _"blocked"_ by failing CI checks which depend on yet unreleased features in the `universal_robot` repository:

- #216, which depends on ros-industrial/universal_robot#380
- #247, which depends on ros-industrial/universal_robot#395

Waiting for a new release of `universal_robot` in order to add those features to the driver doesn't seem very practical, but I'm not sure what alternatives we have.

One option would be to use `UPSTREAM_WORKSPACE=file` and test against a source checkout of `universal_robot` in the tests here. This may bring another set of issues that I may have not thought of, though.

Ideas?

gavanderhoorn, 2019-03-01:
This is a generic issue with repositories that depend on unreleased features.

> One option would be to use `UPSTREAM_WORKSPACE=file` and test against a source checkout of `universal_robot` in the tests here. This may bring another set of issues that I may have not thought of, though.

this is not really an option imo because we'd be building against a moving target, instead of against the released state of a dependency.

What we *could* perhaps do is *add* a build configuration that builds against `kinetic-devel` of `universal_robot` in addition to all the others. In case of PRs against `ur_modern_driver` that are part of a larger set of changes, including `universal_robot`, we could ignore the failures that build against the released `.deb`s and only look at the source build. Only admins would be able to merge those PRs though, as GH would not understand the result of the CI run.

If you'd open a PR to add this, we can discuss the merits of that further.

miguelprada, 2019-03-04:
> this is not really an option imo because we'd be building against a moving target, instead of against the released state of a dependency.

Indeed, but since this is such a "close" dependency and these are basically _devel_ jobs, I thought it may not be such a bad idea. There will eventually be another set of _devel_ jobs in the buildfarm when this gets closer to being released, which can be stricter in this regard than the PR checks here.

> What we could perhaps do is add a build configuration that builds against kinetic-devel of universal_robot in addition to all the others. In case of PRs against ur_modern_driver that are part of a [...]

See #269

ipa-nhg, 2019-03-07:
> One option would be to use UPSTREAM_WORKSPACE=file and test against a source checkout of universal_robot in the tests here. This may bring another set of issues that I may have not thought of, though

+1 for the UPSTREAM_WORKSPACE=file option, I think for this repository, at least for now, makes totally sense.

gavanderhoorn, 2019-04-16:
We've merged #269 which should address this.

closed 2019-04-16 by gavanderhoorn