ros-industrial-attic/ur_modern_driver #267: Want to know driver structure, implement force control and get joint torque

DavidYaonanZhu, 2019-03-01:
Is there possible to implement joint torque control of cartesian force control? I also want to get joint torques. Is anyone possible to give the structure of this driver, like using state machine? And how this driver send commands and read states from the robot?

miguelprada, 2019-03-01:
> Is it possible to implement joint torque control or cartesian force control?

There's no way to command joint torques to the UR robot that I'm aware of. You can still try to control end-effector forces by regulating the position/velocity in an admittance control loop. You'll likely need an external force sensor if you're working with a CB series robot, though, not so sure about the new e-series.

> I also want to get joint torques from the robot.

You can only get joint currents, which are already published part of the `joint_state`. Not sure how useful these are, in any case.

> Is anyone possible to give the structure of this driver, like using state machine?
> And how this driver send commands and read states from the robot?

I believe you'll need to read the source code. I don't think there is any detailed documentation of the driver's structure.

closed 2019-03-01 by miguelprada

gavanderhoorn, 2019-03-01:
For the `kinetic-devel` refactor, @Zagitta actually wrote a thesis that includes information on the design: [Real-time Control of Robotswith ROS](http://production.datastore.cvt.dk/oafilestore?oid=598aeb6f5010df755c717ee7&targetid=598aeb6f5010df755c717ee9).

DavidYaonanZhu, 2019-03-02:
Thank You very much!