ros-industrial-attic/ur_modern_driver #271: Stop sending speedj()

Chunting, 2019-03-18:
Hi, 
My environment is Ubuntu 14.04+ROS Indigo+CB 3.8
In the ros control mode, it seems that no matter there is a new command or not, the following line always sends speedj() command to the real robot.
[hardware_interface_->write();](https://github.com/ros-industrial/ur_modern_driver/blob/3f7cbbf815ff3b736f99462d42e33e53b63c7629/src/ur_ros_wrapper.cpp#L599)

Is it possible to stop sending anything if there is no new command?
I printed out the command, it is not absolute zero, which causes unexpected oscillation of the robot.

Thanks for your kind help.
Chunting

gavanderhoorn, 2019-03-18:
No, there is no way to "stop sending speedj()", as the `ros_control` mode consists of a closed-loop control setup. So even if there are no new setpoints provided, control is still run to maintain current state.

What could be done however would be to implement some sort of low-pass filter / truncation which would essentially "zero-out" very small velocities to avoid the oscillatory behaviour that you've noticed. I've seen that used in other `hardware_interface`s.

> Another question, how does it get the new command [joint_velocity_command_[i]](https://github.com/ros-industrial/ur_modern_driver/blob/3f7cbbf815ff3b736f99462d42e33e53b63c7629/src/ur_hardware_interface.cpp#L159)?

You link to the current `master` version. Are you using `master` or `kinetic-devel`? You should be using `kinetic-devel`.

Chunting, 2019-03-19:
Thanks for your quick reply @gavanderhoorn
1. Would you please provide further information about the filter?
I've tried sending the absolute zero speedj(.0,.0,.0,.0,.0,.0, ) command explicitely, but it did not work at all.
What I want is to avoid oscillatory behaviour from the robot, not just post/pre-processing the data.

2. I'm working on the master branch (Indigo). I started my project a couple of years ago (2016), the kinetic-devel did not exist then.
No idea how much work I need to do to upgrade Ubuntu and reset up my work on kinetic.

Thanks for your kind help.
Chunting

miguelprada, 2019-03-19:
As @gavanderhoorn pointed out, even if there's no new (I'm guessing trajectory) setpoint, the controller is still closing the loop at every cycle.

One alternative you have, @Chunting, is to stop the controller when you're not sending new trajectories. We've done this in different projects and works flawlessly. As a bonus, it'll enable sending custom URScript commands, which isn't the case while the controller is running, since they'll be constantly preempted by the `speedj` commands sent at 125Hz.

Regarding the specific branch to use, the refactored version in `kinetic-devel` should be a drop-in replacement of the one in `master`. You should definitely try it, since it's the most up to date version.

Chunting, 2019-03-20:
Thanks for your help @miguelprada.
I've taken your advice and  tried kinetic-devel. It seems that it is not necessary to upgrade to Ubuntu 16.06. 
But something does not work, I can not connect the real robot.
would you please give me some tips about the configuration? launch file and CMakefile.
This is the current launch file I 'm using.

`
  <group ns="$(arg controller_name_left)">
    <remap from="joint_states" to="/joint_states"/>
    
    <node name="ur_hardware_interface" pkg="ur_modern_driver" type="ur_driver" output="screen" launch-prefix="">
      <param name="robot_ip_address" type="str" value="$(arg ip_left)"/>
      <param name="min_payload" type="double" value="$(arg min_payload)"/>
      <param name="max_payload" type="double" value="$(arg max_payload)"/>
      <param name="max_velocity" type="double" value="$(arg max_velocity_left)"/>
      <param name="use_ros_control" type="bool" value="True"/>
      <param name="prefix" value="$(arg prefix_left)" />
      <param name="base_frame" type="str" value="$(arg base_frame)"/>
      <param name="tool_frame" type="str" value="$(arg left_tool_frame)"/>
    </node>
    
    
    <rosparam file="$(find dual_arm_robot_driver)/config/left_controllers.yaml" command="load"/>

    
    <node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="left_joint_state_controller left_vel_based_pos_traj_controller --shutdown-timeout 1" />

    
    <node name="ros_control_controller_manager" pkg="controller_manager" type="controller_manager" respawn="false"
    output="screen" args="load left_vel_based_admittance_traj_controller" />

    
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"/>
  </group>

  
  <group ns="$(arg controller_name_right)">
    <remap from="joint_states" to="/joint_states"/>
    
    <node name="ur_hardware_interface" pkg="ur_modern_driver" type="ur_driver" output="screen" launch-prefix="">
      <param name="reverse_port" type="int" value="5002"/>
      <param name="robot_ip_address" type="str" value="$(arg ip_right)"/>
      <param name="min_payload" type="double" value="$(arg min_payload)"/>
      <param name="max_payload" type="double" value="$(arg max_payload)"/>
      <param name="max_velocity" type="double" value="$(arg max_velocity_right)"/>
      <param name="use_ros_control" type="bool" value="True"/>
      <param name="prefix" value="$(arg prefix_right)" />
      <param name="base_frame" type="str" value="$(arg base_frame)"/>
      <param name="tool_frame" type="str" value="$(arg right_tool_frame)"/>
    </node>
    
    
    <rosparam file="$(find dual_arm_robot_driver)/config/right_controllers.yaml" command="load"/>

    
    <node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="right_joint_state_controller right_vel_based_pos_traj_controller --shutdown-timeout 1" />

    
    <node name="ros_control_controller_manager" pkg="controller_manager" type="controller_manager" respawn="false"
    output="screen" args="load right_vel_based_admittance_traj_controller" />

    
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"/>
  </group>`

I really appreciate your help.

Thanks,
Chunting

miguelprada, 2019-03-20:
Too many unknowns to know why your specific setup doesn't work. Can you try one of the provided launchfiles/configurations, see if that works, and then proceed from there?

gavanderhoorn, 2019-03-20:
In addition to @miguelprada's comment I would say that whenever reporting issues *always* include log output @Chunting. How are we supposed to know what is going on with your system if all you write is "it doesn't appear to work"?

Chunting, 2019-03-25:
Thanks for your kind reminder. @gavanderhoorn 
Now I'm working on ROS Kinetic + devel-kinetic + Ubuntu 16.04 + CB3.8.
I tried the ros control mode, and here is the output message, I highlighted the errors.

> 
> chunting@chunting-ThinkPad-T460p:~/catkin_ws$ roslaunch ur_modern_driver ur5_ros_control.launch
> ... logging to /home/chunting/.ros/log/ab98bb98-4ec9-11e9-b11d-44850011896a/roslaunch-chunting-ThinkPad-T460p-16116.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://chunting-ThinkPad-T460p:35531/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /force_torque_sensor_controller/publish_rate: 125
>  * /force_torque_sensor_controller/type: force_torque_sens...
>  * /hardware_control_loop/loop_hz: 125
>  * /hardware_interface/joints: ['shoulder_pan_jo...
>  * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
>  * /joint_group_vel_controller/type: velocity_controll...
>  * /joint_state_controller/publish_rate: 125
>  * /joint_state_controller/type: joint_state_contr...
>  * /pos_based_pos_traj_controller/action_monitor_rate: 10
>  * /pos_based_pos_traj_controller/constraints/elbow_joint/goal: 0.1
>  * /pos_based_pos_traj_controller/constraints/elbow_joint/trajectory: 0.1
>  * /pos_based_pos_traj_controller/constraints/goal_time: 0.6
>  * /pos_based_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
>  * /pos_based_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.1
>  * /pos_based_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
>  * /pos_based_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.1
>  * /pos_based_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
>  * /pos_based_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
>  * /pos_based_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.1
>  * /pos_based_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
>  * /pos_based_pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.1
>  * /pos_based_pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
>  * /pos_based_pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.1
>  * /pos_based_pos_traj_controller/joints: ['shoulder_pan_jo...
>  * /pos_based_pos_traj_controller/state_publish_rate: 125
>  * /pos_based_pos_traj_controller/stop_trajectory_duration: 0.5
>  * /pos_based_pos_traj_controller/type: position_controll...
>  * /robot_description: <?xml version="1....
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
>  * /ur_hardware_interface/base_frame: base
>  * /ur_hardware_interface/max_payload: 3.0
>  * /ur_hardware_interface/max_velocity: 10.0
>  * /ur_hardware_interface/min_payload: 0.0
>  * /ur_hardware_interface/prefix: 
>  * /ur_hardware_interface/reverse_port: 50001
>  * /ur_hardware_interface/robot_ip_address: 192.168.1.101
>  * /ur_hardware_interface/servoj_gain: 750.0
>  * /ur_hardware_interface/shutdown_on_disconnect: True
>  * /ur_hardware_interface/tool_frame: tool0_controller
>  * /ur_hardware_interface/use_ros_control: True
>  * /vel_based_pos_traj_controller/action_monitor_rate: 10
>  * /vel_based_pos_traj_controller/constraints/elbow_joint/goal: 0.1
>  * /vel_based_pos_traj_controller/constraints/elbow_joint/trajectory: 0.1
>  * /vel_based_pos_traj_controller/constraints/goal_time: 0.6
>  * /vel_based_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
>  * /vel_based_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.1
>  * /vel_based_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
>  * /vel_based_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.1
>  * /vel_based_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
>  * /vel_based_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
>  * /vel_based_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.1
>  * /vel_based_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
[... 49 lines cut ...]
> setting /run_id to ab98bb98-4ec9-11e9-b11d-44850011896a
> process[rosout-1]: started with pid [16143]
> started core service [/rosout]
> process[ur_hardware_interface-2]: started with pid [16160]
> process[ros_control_controller_spawner-3]: started with pid [16161]
> process[ros_control_controller_manager-4]: started with pid [16162]
> process[robot_state_publisher-5]: started with pid [16175]
> [INFO] [1553496369.970259]: Controller Spawner: Waiting for service controller_manager/load_controller
> [INFO] [1553496370.274897]: Controller Spawner: Waiting for service controller_manager/switch_controller
> [INFO] [1553496370.276112]: Controller Spawner: Waiting for service controller_manager/unload_controller
> [INFO] [1553496370.277394]: Loading controller: joint_state_controller
> [INFO] [1553496370.288948]: Loading controller: force_torque_sensor_controller
**> [ERROR] [1553496370.292089301]: Could not load controller 'force_torque_sensor_controller' because controller type 'force_torque_sensor_controller/ForceTorqueSensorController' does not exist.
> [ERROR] [1553496370.292134016]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
> Loaded pos_based_pos_traj_controller
> [ERROR] [1553496370.552956517]: Could not load controller 'joint_group_vel_controller' because controller type 'velocity_controllers/JointGroupVelocityController' does not exist.
[... cut ...]
> log file: /home/chunting/.ros/log/ab98bb98-4ec9-11e9-b11d-44850011896a/ros_control_controller_manager-4*.log
> [ERROR] [1553496371.293158]: Failed to load force_torque_sensor_controller**
> [INFO] [1553496371.294111]: Loading controller: vel_based_pos_traj_controller
> [INFO] [1553496371.560824]: Controller Spawner: Loaded controllers: joint_state_controller, vel_based_pos_traj_controller
> [INFO] [1553496371.568657]: Started controllers: joint_state_controller, vel_based_pos_traj_controller
>

Chunting, 2019-03-25:
Here is the controller list. It seems strange to me that the driver did not load **force_torque_sensor_controller** and **joint_group_vel_controller.**

> chunting@chunting-ThinkPad-T460p:~/catkin_ws$ rosservice call controller_manager/list_controller_types
> types: [effort_controllers/JointEffortController, effort_controllers/JointGroupEffortController,
>   effort_controllers/JointGroupPositionController, effort_controllers/JointPositionController,
>   effort_controllers/JointTrajectoryAdmittanceController, effort_controllers/JointTrajectoryController,
>   effort_controllers/JointVelocityController, joint_state_controller/JointStateController,
>   my_cart_controller_pkg/MyCartControllerPlugin, my_controller_pkg/MyControllerPlugin,
>   pos_vel_acc_controllers/JointTrajectoryController, pos_vel_controllers/JointTrajectoryController,
>   position_controllers/JointTrajectoryAdmittanceController, position_controllers/JointTrajectoryController,
>   velocity_controllers/JointTrajectoryAdmittanceController, velocity_controllers/JointTrajectoryController]
> base_classes: ['controller_interface::ControllerBase', 'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
>   'controller_interface::ControllerBase']

gavanderhoorn, 2019-03-25:
Do you have those packages installed?

What is the output of `dpkg -l | grep force-torque` and `dpkg -l | grep velocity-controllers`?

I noticed you have this in your `list_controllerer_types` output:

```
position_controllers/JointTrajectoryAdmittanceController
```

that is probably a controller from AndyZe. Did you install some things from source?

gavanderhoorn, 2019-03-25:
But tbh, this is getting off-topic wrt your initial question (and more a subject for ROS Answers).

---

I think @miguelprada answered your original question:

> One alternative you have, @Chunting, is to stop the controller when you're not sending new trajectories. We've done this in different projects and works flawlessly. As a bonus, it'll enable sending [...]

If you have the option, I'd suggest to look into controller switching.

Anything else will require changes to the sources.

gavanderhoorn, 2019-03-25:
I'll close this, assuming that your initial question was addressed.

If you feel this is not the case, or encounter problems using controller switching, please re-open or open a new issue.

Feel free to keep commenting on this issue of course.

closed 2019-03-25 by gavanderhoorn