ros-industrial-attic/ur_modern_driver #274: Action client not connected

ri-ceres, 2019-03-27:
Following installation instructions in a new workspace, when trying to run `roslaunch ur5_e_moveit_config ur5_e_moveit_planning_execution.launch
`, the action will not connect without modifying `action_server.cpp` to include the relative path `vel_based_pos_traj_controller` (or `pos_based_pos_traj_controller`).

```
[ERROR] [1553722932.846096862]: Action client not connected: /vel_based_pos_traj_controller/follow_joint_trajectory
```

In context:
```
[ INFO] [1553722920.267220006]: Loading robot model 'ur5e'...
[ WARN] [1553722920.267284107]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1553722920.267305217]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1553722920.357475952]: Starting planning scene monitor
[ INFO] [1553722920.359920105]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1553722920.360577737]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ WARN] [1553722920.845698257]: Waiting for /vel_based_pos_traj_controller/follow_joint_trajectory to come up
[ INFO] [1553722925.362368950]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-melodic-moveit-ros-planning-0.10.8/planning_scene_monitor/src/planning_scene_monitor.cpp:495
[ INFO] [1553722925.657439607]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ WARN] [1553722926.845889703]: Waiting for /vel_based_pos_traj_controller/follow_joint_trajectory to come up
[ERROR] [1553722932.846096862]: Action client not connected: /vel_based_pos_traj_controller/follow_joint_trajectory
[ INFO] [1553722932.855309229]: Returned 0 controllers in list
[ INFO] [1553722932.865572089]: Trajectory execution is managing controllers

```

`universal_robot/ur5_e_moveit_config/config.yml` looks like:
```
controller_list:
 - name: /vel_based_pos_traj_controller #or /pos_based_pos_traj_controller 
   action_ns: follow_joint_trajectory
   type: FollowJointTrajectory
   default: true
   joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
```

Edit to `ur_modern_driver/src/ros/action_server.cpp` lines 4-16
```
ActionServer::ActionServer(ActionTrajectoryFollowerInterface& follower, std::vector<std::string>& joint_names,
                           double max_velocity)
  : as_(nh_, "pos_based_pos_traj_controller/follow_joint_trajectory", boost::bind(&ActionServer::onGoal, this, _1),
        boost::bind(&ActionServer::onCancel, this, _1), false)
  , joint_names_(joint_names)
  , joint_set_(joint_names.begin(), joint_names.end())
  , max_velocity_(max_velocity)
  , interrupt_traj_(false)
  , has_goal_(false)
  , running_(false)
  , follower_(follower)
  , state_(RobotState::Error)
{
```

Seems like this should not be the intended way to set the path. Is this an issue with using the `kinetic-devel` on ROS melodic? 

Ubuntu 18.04
ROS Melodic
UR5e firmware 5.2.1

gavanderhoorn, 2019-03-28:
This could be related to https://github.com/ros-industrial/ur_modern_driver/pull/216#issuecomment-472963812.

In any case: we'd need to see the output of the controller spawner as that actually loads the controllers.

Please also provide the output of `rostopic list`.

Edits to source code for topic names is *never* needed. This is most likely a configuration issue (on the `ros_control` or the MoveIt side).

gavanderhoorn, 2019-03-28:
Oh and I assumed it, but just to make sure: you are using the version of the driver from #216, correct?

miguelprada, 2019-03-28:
It looks like you should modify the sample MoveIt configuration so that the `move_group` node knows which namespace to look into when trying to send the trajectories to the robot driver. Look into `controllers.yaml` inside the `ur5_e_moveit_config`. Or even better, create a brand new MoveIt config package for your specific setup using the setup assistant.

Also, this is not really a problem with the driver _per-se_, but rather an integration question, and therefore [ROS Answers](https://answers.ros.org/questions/) would be a better place to ask for help about it.

gavanderhoorn, 2019-03-28:
@miguelprada: @ri-ceres showed the `config.yml` from `ur5_e_moveit_config` so I assumed he is/was using a configuration that is known-good-to-work.

And just noticed this line in the log:

```
Waiting for /vel_based_pos_traj_controller/follow_joint_trajectory to come up
```

I'm not sure why @ri-ceres then shows this in his edit of `action_server.cpp`:

```c++
as_(nh_, "pos_based_pos_traj_controller/follow_joint_trajectory", [..]
```

those are two different controllers.

gavanderhoorn, 2019-03-28:
Btw @ri-ceres: where did the file `config.yml` come from? It's not one of the files in [ur5_e_moveit_config/config](https://github.com/ros-industrial/universal_robot/tree/5d9f697988d68cf499be9661298da9786c351b8d/ur5_e_moveit_config/config).

miguelprada, 2019-03-28:
There's no such thing as a `config.yaml`. The contents shown in the original comment suggest he's refering to `ur5_e_moveit_config/config/controllers.yaml`, and that's one of the reasons I explicitly mentioned it in my comment.

@ri-ceres how are you starting the driver node? The correct namespace for MoveIt to attempt to connect to is different depending on whether you are using the `ros_control` mode or not.

ri-ceres, 2019-03-28:
The `config.yml` came from the _Usage_ section in the [README.md](https://github.com/ros-industrial/ur_modern_driver/blob/master/README.md#usage) for `ros-industrial/ur_modern_driver` where it says "To use ros_control together with MoveIt...". 

I tried following those instructions as well as using the default `config.yml` that you linked above([ur5_e_moveit_config/config](https://github.com/ros-industrial/universal_robot/tree/5d9f697988d68cf499be9661298da9786c351b8d/ur5_e_moveit_config/config)]. I get the same result except that it isn't looking for the prefix `pos_based_pos_traj_controller` anymore:

```
[ INFO] [1553722468.842190626]: Loading robot model 'ur5e'...
[ WARN] [1553722468.842270988]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1553722468.842298810]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1553722468.933376966]: Starting planning scene monitor
[ INFO] [1553722468.935623766]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1553722468.936192068]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ WARN] [1553722469.435401928]: Waiting for /follow_joint_trajectory to come up
[ INFO] [1553722473.937926359]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-melodic-moveit-ros-planning-0.10.8/planning_scene_monitor/src/planning_scene_monitor.cpp:495
[ INFO] [1553722474.209165112]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ WARN] [1553722475.435560190]: Waiting for /follow_joint_trajectory to come up
[ERROR] [1553722481.435762430]: Action client not connected: /follow_joint_trajectory
[ INFO] [1553722481.449385025]: Returned 0 controllers in list
[ INFO] [1553722481.459512214]: Trajectory execution is managing controllers

```

As for the different controller in `action_server.cpp` you pointed out [above](https://github.com/ros-industrial/ur_modern_driver/issues/274#issuecomment-477559857), that was my mistake when creating this issue. It should have been `vel_based_pos_traj_controller/follow_joint_trajectory` (although I also tried using the `pos_based_pos_traj_controller` at one point). I just now checked for consistency across files locally to make sure that the version I'm using didn't have this error, and everything looks right. 

Regarding the suggestion by @miguelprada that I need to create a custom moveit package and that my issue isn't appropriate for this repo, I disagree. I'm not using anything that would require customization, and assumed that there should be a known to work configuration as @gavanderhoorn mentioned.

When you say I should be using [#216](https://github.com/ros-industrial/ur_modern_driver/pull/216), should I be using @miguelprada's [ur-e branch](https://github.com/ros-industrial/ur_modern_driver/pull/216#issuecomment-472963812)? Figuring out which UR forks and branches to be using for both the e-series and ROS melodic seems a little convoluted at present.

ri-ceres, 2019-03-28:
To follow up on this, I just tried the [ur-e branch in @miguelprada's fork](https://github.com/miguelprada/ur_modern_driver/tree/ur-e) and that seems to work on our robot. 

Thanks @gavanderhoorn  and @miguelprada for helping direct me to the right place to look for this.

miguelprada, 2019-03-29:
I'm surprised that using `ur-e` branch solves this, since there's nothing there that should affect the namespaces for the action servers for the joint trajectory actions.

What's clear, anyways, is that as @gavanderhoorn suggested, you shouldn't need to modify the source code of `action_server.cpp`, but rather change the moveit config in `controllers.yaml` to point to the correct action server namespace.

Can this issue be closed then?

ri-ceres, 2019-03-29:
Yes, thanks.

closed 2019-03-29 by ri-ceres