ros-industrial-attic/ur_modern_driver #276: kinetic-devel branch does not contain copyright statements in files

fmauch, 2019-03-29:
Currently the kinetic-devel branch is missing copyright statements inside the files, which to my understanding is quite common in open-source projects.

To be clear, I'm talking about such statements as on the master branch:
```
/*
 * ur_communication.h
 *
 * Copyright 2015 Thomas Timm Andersen
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */
```

Is this intentional or is this really missing?

gavanderhoorn, 2019-03-29:
I can't speak for @Zagitta , but as far as I understand it this is intentional *and* and oversight: it is quite common in ROS projects to only find a license file in the root of the repository and the assumption that all files in that repository fall under that license.

But I understand that this complicates reuse of such files in projects that would mix-and-match files from several repositories.

I've seen statements in readmes to the effect of "files X, Y and Z come from the P project and fall under the Q license". But that quickly becomes cumbersome and unmaintainable. Apart from that, authorship information of the file itself is lost that way.

Concluding: we should add license headers to the files.

I'll submit a PR.

gavanderhoorn, 2019-03-29:
Thanks for bringing this to our attention @fmauch.

gavanderhoorn, 2019-03-29:
Please see #277 which should resolve this.

gavanderhoorn, 2019-04-01:
@fmauch: I've just merged #277, which should resolve this situation.

commit 1f5d04b9 (2019-04-01): Merge pull request #277 from gavanderhoorn/add_license_headers

Add license headers (fix #276)
closed 2019-04-01 by gavanderhoorn