ros-industrial-attic/ur_modern_driver #280: Potentially unsafe static_cast of double to int in SetIO service handler

gavanderhoorn, 2019-04-03:
@miguelprada wrote (in https://github.com/ros-industrial/universal_robot/pull/415#issuecomment-478891361):
> It makes me think whether `ur_modern_driver` should use `std::lround` instead of `static_cast<uint8_t>` to [transform tool voltages](https://github.com/ros-industrial/ur_modern_driver/blob/7fae9f236e9557eb09e5e6554abe54da9de8f24a/include/ur_modern_driver/ros/io_service.h#L53) just to be safer.

gavanderhoorn, 2019-04-16:
Labelled as `help wanted` as this would be a good first issue to work on.

reidchristopher, 2019-07-02:
Seems like a change would be unnecessary. A message sent should use the enumerations for 0, 12, and 24, meaning there should never be any values like 11.99 or 23.99 that would give reason for that change. Please do correct me if I misunderstand.

gavanderhoorn, 2019-07-04:
> A message sent should use the enumerations for 0, 12, and 24, meaning there should never be any values like 11.99 or 23.99 that would give reason for that change.

I guess the important bit here is "should use the enumerations".

Fractional values are not expected, I agree, but perhaps it might make sense to -- instead of simply `static_cast`ing the value to an integer -- be defensive and check that one of the acceptable values is being passed in.

Prior to #415 there was no set of constants that could be used, so clients could be specifying any value.

gavanderhoorn, 2019-07-05:
@reidchristopher: would this be something you see opportunity to work on? Otherwise I'm going to remove the `wrid19` label from the issue and remove it from the board.

reidchristopher, 2019-07-05:
The value is actually being checked by  <code>URCommander::setToolVoltage</code>  and will return that the service failed if the value is not exactly 0, 12, or 24. At the end of the day, I have to admit it should do no harm to change the casting to rounding, other than the one off chance that individuals believe they are setting the voltage to a slightly lower value than they are.

reidchristopher, 2019-07-05:
Within the last few seconds, I realized the checking logic is flawed and will always return false. That at the very least needs to be fixed and I would be happy to do so

reidchristopher, 2019-07-05:
Fixed the logic and added the rounding in #327