ros-industrial-attic/ur_modern_driver #282: Create a ur_modern_driver_tools package

gavanderhoorn, 2019-04-05:
To host contributions like the convenience functions that were originally part of #247.

It should host scripts (and perhaps nodes) that are not directly part of the driver, but would still be useful to keep close to the driver.

Making this a separate package would allow the use of such tools without having to install/build the driver itself as well.

gavanderhoorn, 2019-04-05:
@felixvd added [wait_for_program.h ](https://gist.github.com/felixvd/4d5db1c297a3435cabb40c857eb6ead5) in #247, which would be a good first candidate for the package.

gavanderhoorn, 2019-04-05:
[test_move.py](https://github.com/ros-industrial/ur_modern_driver/blob/195f5e937af6459f8808df3f021923614e761413/test_move.py) would be another candidate (not this version specifically though).

miguelprada, 2019-04-05:
Not particularly against this, but we need to choose when to do it wisely, since moving current content into a sub-folder will conflict with all open PRs.

gavanderhoorn, 2019-11-23:
@felixvd: it may make sense to see whether your convenience function could be updated to be used with `ur_robot_driver`. If it can, please open an issue on the `ur_robot_driver` issue tracker to discuss where it should be hosted.

If you'd be interested of course.

closed 2019-11-23 by gavanderhoorn