ros-industrial-attic/ur_modern_driver #285: topic robot_status

JoeryTemmink, 2019-04-08:
I want to get the status of the robot Ur10e. For example when the robot is in protected stop i want to know this. So i subscribe on the topic robot_status and receive this: 
`----
header: 
  seq: 59
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
mode: 
  val: 2
e_stopped: 
  val: 0
drives_powered: 
  val: 1
motion_possible: 
  val: 1
in_motion: 
  val: -1
in_error: 
  val: 0
error_code: 0
---`
But i can't find for example what mode 2 (or other index) means. Same with the error_code. Or do i need to use something else of this?

gavanderhoorn, 2019-04-08:
A `rostopic type /robot_status` should show you that the topic type is `industrial_msgs/RobotStatus`.

You could then either use `rosmsg show industrial_msgs/RobotStatus`, or use the documentation on the ROS wiki: [industrial_msgs/RobotStatus](http://docs.ros.org/melodic/api/industrial_msgs/html/msg/RobotStatus.html).

gavanderhoorn, 2019-04-08:
Closing, as this is not an actionable issue, but a question about the use of the package / interfaces.

Feel free to keep commenting on this issue of course @JoeryTemmink.

closed 2019-04-08 by gavanderhoorn