ros-industrial-attic/ur_modern_driver #289: Unused min/max_payload parameters

miguelprada, 2019-04-10:
This package's launch files expose `min_payload` and `max_payload` arguments, which are used to set equally named parameters, but those parameters are not used anywhere.

gavanderhoorn, 2019-04-10:
Those were supported in the `master` version (which got it from `ur_driver`), but apparently @Zagitta didn't implement the checks:

https://github.com/ros-industrial/ur_modern_driver/blob/77fa08ae9c846344310d3b50824a7affdc3eda47/include/ur_modern_driver/ros/io_service.h#L65-L70

gavanderhoorn, 2019-04-10:
I added `help wanted` as this could be a good first issue for someone to tackle.

gavanderhoorn, 2019-07-03:
Did you make any progress here @ben-greenberg?

ben-greenberg, 2019-07-03:
I have made progress but some further work and testing is still needed. I will be continuing my work on this.

gavanderhoorn, 2019-07-17:
I'm going to unassign this @ben-greenberg.

You're welcome to keep working on it of course.