ros-industrial-attic/ur_modern_driver #291: Unable to send trajectories after switch to Manual mode

natharos, 2019-04-16:
ROS Kinetic
Ubuntu 16.04
UR10e robot

When the robot is set to Manual mode and then back to Remote mode with the teach pendant, ROS states that it is reconnected to the robot, but new trajectories are not received and executed anymore. ROS is stuck at "Uploading trajectory program to robot". How can I fix this?

> [ INFO] [1555396555.040235408]: Received request to compute Cartesian path
> [ INFO] [1555396555.042064984]: Attempting to follow 5 waypoints for link 'ee_link' using a step of 0.010000 m and jump threshold 10.000000 (in global reference frame)
> [ INFO] [1555396555.132936089]: Computed Cartesian path with 67 points (followed 100.000000% of requested trajectory)
> [ INFO] [1555396555.190454640]: Received new trajectory execution service request...
> [ INFO] [1555396555.207087763]: Received new goal
> [ INFO] [1555396555.207201327]: Trajectory received and accepted
> [ INFO] [1555396555.207331660]: Translating trajectory
> [ INFO] [1555396555.207396334]: Executing trajectory with 67 points and duration of 2.806s
> [ INFO] [1555396555.207451057]: Attempting to start follower 0x268f0c0
> [ INFO] [1555396555.207500532]: Uploading trajectory program to robot
> [ INFO] [1555396558.018604466]: Trajectory executed successfully
> [ INFO] [1555396558.043807630]: Completed trajectory execution with status SUCCEEDED ...
> [ INFO] [1555396558.044715104]: Execution completed: SUCCEEDED
> [ WARN] [1555396560.375096315]: Failed to read from stream, reconnecting in 1 seconds...
> [ WARN] [1555396560.446421442]: Failed to read from stream, reconnecting in 1 seconds...
> [ WARN] [1555396561.375202499]: Failed to read from stream, reconnecting in 2 seconds...
> [ WARN] [1555396561.446534172]: Failed to read from stream, reconnecting in 2 seconds...
> [ INFO] [1555396563.375755141]: Setting up connection: 192.168.86.226:30003
> [ INFO] [1555396563.376089679]: Connection established for 192.168.86.226:30003
> [ INFO] [1555396563.446775643]: Setting up connection: 192.168.86.226:30002
> [ INFO] [1555396563.447207264]: Connection established for 192.168.86.226:30002
> [ INFO] [1555396571.029808722]: Received request to compute Cartesian path
> [ INFO] [1555396571.029945972]: Attempting to follow 5 waypoints for link 'ee_link' using a step of 0.010000 m and jump threshold 10.000000 (in global reference frame)
> [ INFO] [1555396571.149077895]: Computed Cartesian path with 68 points (followed 100.000000% of requested trajectory)
> [ INFO] [1555396571.176511582]: Received new trajectory execution service request...
> [ INFO] [1555396571.195561523]: Received new goal
> [ INFO] [1555396571.195650517]: Trajectory received and accepted
> [ INFO] [1555396571.195705772]: Translating trajectory
> [ INFO] [1555396571.195734215]: Executing trajectory with 68 points and duration of 2.619s
> [ INFO] [1555396571.195753971]: Attempting to start follower 0x268f0c0
> [ INFO] [1555396571.195770715]: Uploading trajectory program to robot

miguelprada, 2019-04-16:
Which version of the driver are you using? And how are you starting it (e.g. "default", `ros_control`, low bandwidth mode)?

The issue is likely that even if the data streams coming from the robot are re-connected, the URScript program that's uploaded to the robot in order to control it has been stopped when switching to manual mode and is not re-uploaded/started when switching back to remote mode.

If you're using the `ros_control` mode, re-uploading can be triggered by stopping and starting the controller back. Otherwise, I think your only option is to kill the driver and start it again.

In theory, we could try to detect this situation and upload the program again automatically, but I don't think I'll have much time to look into this in the following weeks. PRs are welcome.

gavanderhoorn, 2019-05-07:
I'm going to close this due to inactivity.

closed 2019-05-07 by gavanderhoorn