ros-industrial-attic/ur_modern_driver #292: UR Driver gives my wrong poses

chandanlal92, 2019-04-16:
Hello,

I am using UR_modern driver with UR-10 Robot to do hand_eye calibration using easy_handeye calibration ros service. The poses which I am recieving from the ur driver seems to be mismatched with the direction.
It gives me
x=-363.23 where my robot panel shows X=363.23
Same cases with Y.Z seems to be correct.

chandanlal92, 2019-04-16:
How to test the current poses?

gavanderhoorn, 2019-04-16:
You'll have to tell us which transform you are looking at.

If you're inspecting `base_link->tool0`, the 180 degree difference is expected.

If you're using `base->tool0`, then you should get the proper poses.

All depending on your toolframe setup also of course.

chandanlal92, 2019-04-16:
Both base->tool0 and base_link->tool0

chandanlal92, 2019-04-16:
Now I am getting correct value after using base->tool0. Thank you for the info.

gavanderhoorn, 2019-04-16:
Please also note: this is not something in `ur_modern_driver`, but in `ur_description` ([here](https://github.com/ros-industrial/universal_robot/blob/2665f4bed158a8962e7ed72c7a83835cb872b59b/ur_description/urdf/ur5.urdf.xacro#L326-L330
)).

closed 2019-04-16 by gavanderhoorn