ros-industrial-attic/ur_modern_driver #293: unable to perform catkin_make (or catkin build) on kinetic devel

JanTromp, 2019-04-16:
Hi, 
I cloned the kinetic-devel ur_modern_driver in my src folder. When i try catkin_make is get the following error:

####
#### Running command: "make pkg ur_modern_driver -j8 -l8" in "/home/mifa/ws_moveit/build"
####
make: *** No rule to make target 'pkg'.  Stop.
Invoking "make pkg ur_modern_driver -j8 -l8" failed

Then my catkin_make stops.

When i try to run the catkin build command i get the following error:

Errors     << ur_modern_driver:make /home/mifa/ws_moveit/logs/ur_modern_driver/build.make.000.log                                                                                                          
In file included from /opt/ros/kinetic/include/hardware_interface/robot_hw.h:35:0,
                 from /opt/ros/kinetic/include/controller_interface/controller_base.h:37,
                 from /opt/ros/kinetic/include/controller_manager/controller_spec.h:41,
                 from /opt/ros/kinetic/include/controller_manager/controller_manager.h:36,
                 from /home/mifa/ws_moveit/src/ur_modern_driver/include/ur_modern_driver/ros/controller.h:20,
                 from /home/mifa/ws_moveit/src/ur_modern_driver/src/ros/controller.cpp:19:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:69:85: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
   static const void callConcatManagers(typename std::vector<T*>& managers, T* result)
                                                                                     ^
In file included from /opt/ros/kinetic/include/hardware_interface/robot_hw.h:35:0,
                 from /opt/ros/kinetic/include/controller_interface/controller_base.h:37,
                 from /opt/ros/kinetic/include/controller_manager/controller_spec.h:41,
                 from /opt/ros/kinetic/include/controller_manager/controller_manager.h:36,
                 from /home/mifa/ws_moveit/src/ur_modern_driver/include/ur_modern_driver/ros/hardware_interface.h:20,
                 from /home/mifa/ws_moveit/src/ur_modern_driver/src/ros/hardware_interface.cpp:19:
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:69:85: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
   static const void callConcatManagers(typename std::vector<T*>& managers, T* result)
                                                                                     ^
In file included from /home/mifa/ws_moveit/src/ur_modern_driver/src/ros/hardware_interface.cpp:19:0:
/home/mifa/ws_moveit/src/ur_modern_driver/include/ur_modern_driver/ros/hardware_interface.h: In constructor ‘VelocityInterface::VelocityInterface(URCommander&, hardware_interface::JointStateInterface&, std::vector<std::__cxx11::basic_string<char> >&, double)’:
/home/mifa/ws_moveit/src/ur_modern_driver/include/ur_modern_driver/ros/hardware_interface.h:75:10: warning: ‘VelocityInterface::max_vel_change_’ will be initialized after [-Wreorder]
   double max_vel_change_;
          ^
/home/mifa/ws_moveit/src/ur_modern_driver/include/ur_modern_driver/ros/hardware_interface.h:74:40: warning:   ‘std::array<double, 6ul> VelocityInterface::prev_velocity_cmd_’ [-Wreorder]
   std::array<double, 6> velocity_cmd_, prev_velocity_cmd_;
                                        ^
/home/mifa/ws_moveit/src/ur_modern_driver/src/ros/hardware_interface.cpp:50:1: warning:   when initialized here [-Wreorder]
 VelocityInterface::VelocityInterface(URCommander &commander, hardware_interface::JointStateInterface &js_interface,
 ^
In file included from /home/mifa/ws_moveit/src/ur_modern_driver/src/ros/mb_publisher.cpp:19:0:
/home/mifa/ws_moveit/src/ur_modern_driver/include/ur_modern_driver/ros/mb_publisher.h:27:38: fatal error: ur_msgs/RobotModeDataMsg.h: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/ur_driver.dir/src/ros/mb_publisher.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/ur_driver.dir/all] Error 2
make: *** [all] Error 2
cd /home/mifa/ws_moveit/build/ur_modern_driver; catkin build --get-env ur_modern_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd 

I am new to ros so i have no idea what i'm doing when i run catkin make or build.

Annyone knows what is making this crash?

Kind regards,
Jan Tromp.

martimorta-wood, 2019-04-16:
Hi Jan, catkin_make should be ran from the workspace's root, maybe that's the problem.

I **strongly recommend** to learn ROS before trying to work with more complex stuff, the best way is following the [tutorials](http://wiki.ros.org/ROS/Tutorials)

gavanderhoorn, 2019-04-16:
There are a few warnings, but this is the only error:

```
In file included from /home/mifa/ws_moveit/src/ur_modern_driver/src/ros/mb_publisher.cpp:19:0:
/home/mifa/ws_moveit/src/ur_modern_driver/include/ur_modern_driver/ros/mb_publisher.h:27:38: fatal error: ur_msgs/RobotModeDataMsg.h: No such file or directory
```

@JanTromp: how have you setup your workspace and installed ROS?

`RobotModeDataMsg` is definitely part of `ur_msgs`, so I'm wondering why you are missing it.

miguelprada, 2019-04-16:
> `RobotModeDataMsg` is definitely part of `ur_msgs`, so I'm wondering why you are missing it.

Yes, but it's probably not in the released binaries until the next sync.

@JanTromp the build should probably succeed if you add `kinetic-devel` branch of `universal_robot` repository to your workspace as well, until the updated version of the binaries is released.

JanTromp, 2019-04-16:
Update:

I redownloaded the universal_robot package (this time made sure i had the kinetic devel branch) en ran catkin build. It succeeded:

**[build] Summary: All 19 packages succeeded!                                    
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    None.**

The warningL /opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:69:85: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
   static const void callConcatManagers(typename std::vector<T*>& managers, T* result)

But i guess i can ignore this warning for now right?

Thank you all for your input!

gavanderhoorn, 2019-04-16:
Yes, you can ignore the warning for now.

As @miguelprada writes: as soon as the next Kinetic sync is out (announced [here](https://discourse.ros.org/c/release/kinetic)) you don't need to have a clone of `universal_robot` in your workspace any more.

gavanderhoorn, 2019-04-16:
I've pinned this as I expect other users to run into this as well -- or at least until the Kinetic sync.

closed 2019-04-16 by gavanderhoorn

JanTromp, 2019-04-16:
Okay so now i tried moving my ur10 with the updated packages but i'm experiencing some disturbing issues:

First of all (a not so disturbing issue): When i open rviz to move the robot i perform the command **roslaunch ur10_moveit_config moveit_rviz.launch config:=true** (as stated in the ur_modern_driver usage) Rviz opens but i dont have the interactive markers to move the robot around.

[image]

Second (and more concerning): when i move plan a path to go from the current position to the predefined "up" position, the planning goes well and it shows me a nice path. Then when i press execute the robot is moving through the "up" position and continues to move to a unknown position. Untill i have to press the emergency stop since the robot will crash into itself.

I dont know if this problem is due to the ur_modern_driver package, universal_robot package or something in moveit or rviz.

gavanderhoorn, 2019-04-16:
Please do not post additional problems on a closed issue.

at the very least open a new issue.

gavanderhoorn, 2019-04-30:
Unpinning as Kinetic [has been synced](https://discourse.ros.org/t/preparing-for-kinetic-sync-2019-04-23/8895/2).