ros-industrial-attic/ur_modern_driver #294: Problems with moving ur10 through moveit/Rviz

JanTromp, 2019-04-16:
Hi,
I had some issues with getting the ur_modern_driver package to succeed when i "catkin build". This problem was fixed as far as my knowledge (thee this issue [#293](https://github.com/ros-industrial/ur_modern_driver/issues/293))

So the problem was solved but when i moved the ur 
So now i tried moving my ur10 with the updated packages but i'm experiencing some disturbing issues:

First of all (a not so disturbing issue): When i open rviz to move the robot i perform the command **roslaunch ur10_moveit_config moveit_rviz.launch config:=true** (as stated in the ur_modern_driver usage) Rviz opens but i dont have the interactive markers to move the robot around.

[image]

Second (and more concerning): when i move plan a path to go from the current position to the predefined "up" position, the planning goes well and it shows me a nice path. Then when i press execute the robot is moving through the "up" position and continues to move to a unknown position. Until i have to press the emergency stop since the robot will crash into itself.

I don't know if this problem is due to the ur_modern_driver package, universal_robot package or something in moveit or rviz.

martimorta-wood, 2019-04-16:
Hi Jan,
Your answer for the second question is in this post https://github.com/ros-industrial/ur_modern_driver/pull/216#pullrequestreview-170898136
For the first question, please do the MoveIt tutorials (after the ROS ones), the answer here https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#step-2-play-with-the-visualized-robots

Please use one issue per question and before posting a new one search for similar issues or use [ROS answers](https://answers.ros.org/questions)

Regards

JanTromp, 2019-04-16:
update:
I noticed the interactive marker thing was caused by me being stupid and launching the same launch file twice :page_facing_up: 
When i launched all the necessary files only once the markers showed and i was able to move the robot using the markers. So this is **FIXED**

As for the second problem. When i moved the robot to any position other than the up position it worked fine. So i think the joint angles of the up position aren't correct or something. Not something to worry about because i won't be using this position. 

As for your comment
"Please use one issue per question and before posting a new one search for similar issues or use ROS answers"
The reason i asked both questions in the same post is because i thought they would be related to eachother, so asking them seperatly wouldn't show the connection to you guys.

Anyways, thanks for you help!

gavanderhoorn, 2019-04-16:
> As for the second problem. When i moved the robot to any position other than the up position it worked fine. So i think the joint angles of the up position aren't correct or something

At what setting do you have the speed slider set on the TP?

JanTromp, 2019-04-17:
> > As for the second problem. When i moved the robot to any position other than the up position it worked fine. So i think the joint angles of the up position aren't correct or something
> 
> At what setting do you have the speed slider set on the TP?

The slider on the teach pendant is det to 100%

closed 2019-04-30 by JanTromp