ros-industrial-attic/ur_modern_driver #295: Save trajectory local on the controller

TSsimon, 2019-04-17:
Hello!
I'm about to use the ur_modern_driver to send scheduled trajectory paths to the controller.
My question is, is it possible to save the URscript commands locally in the / Programs directory so that I can see it using the Teachpendant?

gavanderhoorn, 2019-04-17:
No, that's not supported.

It's also not how the driver works: trajectories are *streamed* to the controller. Not generated and then uploaded.

gavanderhoorn, 2019-04-17:
I'm closing this as it's not an actionable issue.

Feel free to keep commenting here of course.

closed 2019-04-17 by gavanderhoorn

TSsimon, 2019-04-17:
then I misunderstood it. I thought that after the socket connection my trajectories would be sent to the ur and executed locally by a pseudo UR script. After execution the thread would be closed

that would mean I steer the servomotors of the ur directly?

gavanderhoorn, 2019-04-17:
> I thought that after the socket connection my trajectories would be sent to the ur and executed locally by a pseudo UR script.

I'm not sure what a "pseudo UR script" is, but there is a small UR script started on the controller and *that* script accepts position (or velocity) setpoints which it then uses to call appropriate `servoj(..)` or `speedj(..)` commands with.

> After execution the thread would be closed

yes, that does happen. See [here](https://github.com/ros-industrial/ur_modern_driver/blob/77fa08ae9c846344310d3b50824a7affdc3eda47/src/ros/trajectory_follower.cpp#L29-L84) for the program that is uploaded.

> that would mean I steer the servomotors of the ur directly?

well .. "directly" is a relative term. It's probably as direct as the controller allows/supports.