ros-industrial-attic/ur_modern_driver #296: question: set I/O's in test- node (c++ or python)

TSsimon, 2019-04-18:
Hello,
I have a question regarding the driver: I know the driver supports setting I/O's.  
I would like to test the I / O's with a simple node, but unfortunately I do not know how to start to program. 
Does anyone have a test node in c++ or python to set the I / O's?

Thanks !

gavanderhoorn, 2019-05-07:
There is a (very) old script in `ur_driver` ([here](https://github.com/ros-industrial/universal_robot/blob/2665f4bed158a8962e7ed72c7a83835cb872b59b/ur_driver/test_io.py)) that you could perhaps use as an inspiration. Note that it uses a Python module of `ur_driver`, so it's not immediately usable with `ur_modern_driver`.

gavanderhoorn, 2019-05-07:
As this is not an actionable issue with the package in this repository, I'm going to close it.

Please feel free to keep commenting on it of course.

closed 2019-05-07 by gavanderhoorn

gonzalocasas, 2019-05-07:
@TSsimon it's easy to do this via the `/URScript` topic: just publish one message containing a string like "set_digital_out(8, True)" (obviously adjusting it to your needs) and that's all.

gavanderhoorn, 2019-05-07:
> and that's all.

unless you want to use services. In which case using the urscript topic is not going to work.

gonzalocasas, 2019-05-07:
> unless you want to use services

true. One option to build a blocking call (at application level) for setting IOs and waiting until completion is to subscribe to the `/ur_driver/io_states` topic right after publishing the urscript message, and there block until the io you just set turns to the state you expect.

gonzalocasas, 2019-05-07:
but since this question starts to be quite long and is unactionable as mentioned by @gavanderhoorn , it might be best if you post it to [ROS answers](https://answers.ros.org/) instead.

gavanderhoorn, 2019-05-07:
@gonzalocasas: do you have any particular reason to *not* use the `set_io` service?

gonzalocasas, 2019-05-07:
None except ignorance 😛 
I had no idea the driver provides such a service, I didn't see it listed in the README.md nor the wiki, and wrongly assumed there was none provided.