ros-industrial-attic/ur_modern_driver #299: Configurable reverse IP address

gonzalocasas, 2019-04-24:
When running the driver node in a network segment that is NAT bridged to the robot's one, the driver sends the local IP to the UR side to initialize the reverse connection on the reverse IP port, but the IP address that it sends is not reachable from the UR's side of the network.

The issue is basically the same as #111 but there was no proper solution applied there. It was closed because the source code was modified locally to point to the reverse IP, but that's not a sustainable solution. Also, the hint regarding using `host` network mode of Docker is limited to Linux systems, which is not our case.

So I would like to send a pull request to add an argument to the driver that allows manually configuring the IP. The argument is optional, and not setting it would lead to the current behavior, hence totally backwards compatible.

Does it sound like something that matches the project and that would be eventually merged? Or is there any fundamental complaint to the approach?

gavanderhoorn, 2019-04-24:
Yes, if you provide a reasonable PR we can certainly merge it.

Using `ur_modern_driver` through a NAT device is not something I would specifically recommend though, but that is a different matter.

miguelprada, 2019-05-09:
Resolved in #301.

closed 2019-05-09 by miguelprada