ros-industrial-attic/ur_modern_driver #300: Unable to run a test program for the ur 10e

aravindmemana, 2019-04-24:
I have tried the roslaunch command and the test programme
roslaunch ur_bringup ur10_bringup.launch robot_ip:=IP_OF_THE_ROBOT
rosrun ur_driver test_move.py

The current status shows programming the robot at the IP and when i run the test program,it shows waiting for the server to respond.
[image]
[image]

Is this a common issue?Also when i run the modern command for the UR robot,it doesnt detect the command.Can anyone please help me out?
[image]

gavanderhoorn, 2019-04-24:
First: please don't post screenshots of terminals. It's all text, so copy-paste the text into your issue description (and format it properly, using three backticks).

Please update your initial post to remove the screenshots.

gavanderhoorn, 2019-04-24:
Having said that: you cannot use `ur_bringup` with `ur_modern_driver`. So you're currently not using `ur_modern_driver`, but `ur_driver`, which is the old, deprecated and incompatible driver component that is not expected to work with anything but CB1 and CB2 controllers.

For e-series compatibility, you'll have to check out the candidate PR #216. Anything else will not work.

aravindmemana, 2019-04-25:
I have cloned the repository and launched the modern driver
  roslaunch ur_modern_driver ur10e_bringup.launch robot_ip:=192.168.1.100

 Error message
  ur10e_bringup.launch is neither a launch file in package ur_modern_driver nor is ur_modern_driver a launch file name
The traceback for the exception was written to the log file.

How can i launch a driver for the robot?.

gonzalocasas, 2019-04-25:
You should follow the standard steps required to run any package that has been cloned from source. Please refer to ROS documentation on setting up your catkin workspace for that, but it basically comes down to: 1) cloning, 2) installing dependencies, 3) compiling, 4) sourcing the workspace:

```
git clone whatever_your_url_might_be
rosdep install -y --from-paths . --ignore-src --rosdistro ${ROS_DISTRO}
catkin_make   # or catkin build if that's your cup of tea
source ~/catkin_ws/devel/setup.bash    # replace with the correct path to your workspace
```

gavanderhoorn, 2019-04-26:
As @gonzalocasas writes, this could be an issue with how you managed your Catkin workspace.

Can you tell us which commands you've used *exactly*?

gavanderhoorn, 2019-04-26:
As right now this doesn't appear to be an issue with the package in this repository, but more a problem with setup / population / use of the build environment, I'm going to close this.

If it turns out this is actually a problem with `ur_modern_driver` itself, we'll re-open.

closed 2019-04-26 by gavanderhoorn

aravindmemana, 2019-04-29:
I have followed the steps mentioned by gonzalocasas.Still facing the same issue.Im new to the platform and trying some means to get the robot running.Thanks

gavanderhoorn, 2019-04-29:
@gavanderhoorn wrote:
> Can you tell us which commands you've used _exactly_?