ros-industrial-attic/ur_modern_driver #302: UR will not stop at the target position unless speed is set to 100%

xarthurx, 2019-04-26:
Using MoveIt to compute trajectory and execute.

If I set the speed to `100%`, no problem. (set on the panel with the slider)
If I set the speed to `n < 100%`, robot will not stop at the target location, but continues to move and slowly stop little by little.

Is this a bug in the driver?

gavanderhoorn, 2019-04-26:
The driver will assume that motion commands take a certain amount of time. By changing the speed slider value, you change that time, but unilaterally: MoveIt doesn't get notified, nor do the controllers.

This is the likely cause of the "desync" that you observed.

> Is this a bug in the driver?

No. At the moment this is expected behaviour.

You could see whether configuring the same velocity and acceleration scaling in your MoveIt plan request improves the situation, but especially the acceleration scaling will not be trivial to get just right.

gavanderhoorn, 2019-04-26:
I'm going to close this as I'm going to assume that we've sufficiently dealt with the issue you report and that it's not something we can change in the driver for now.

Feel free to keep commenting on the issue of course.

closed 2019-04-26 by gavanderhoorn

slessans, 2019-07-01:
this should be documented in the README