ros-industrial-attic/ur_modern_driver #303: how does robotcommand send to robot?

itfanr, 2019-04-30:
I have read the code on kinetic-devel branch and I am confused about the code :

The server with 50001 port reads the packets data and deals with it:

```
bool LowBandwidthTrajectoryFollower::execute(const std::array<double, 6> &positions,
                                             const std::array<double, 6> &velocities, double sample_number,
                                             double time_in_seconds)
{
  if (!running_)
    return false;

  std::ostringstream out;

  out << "(";
  out << sample_number << ",";
  for (auto const &pos : positions)
  {
    out << pos << ",";
  }
  for (auto const &vel : velocities)
  {
    out << vel << ",";
  }
  out << time_in_seconds << ")\r\n";

  // I know it's ugly but it's the most efficient and fastest way
  // We have only ASCII characters and we can cast char -> uint_8
  const std::string tmp = out.str();
  const char *formatted_message = tmp.c_str();
  const uint8_t *buf = (uint8_t *)formatted_message;

  size_t written;
  LOG_DEBUG("Sending message %s", formatted_message);

  return server_.write(buf, strlen(formatted_message) + 1, written);
}
``` 

In URServer class:
```
bool URServer::write(const uint8_t* buf, size_t buf_len, size_t& written)
{
  return client_.write(buf, buf_len, written);
}
```

How does packet send to the robot? I can only find the code above. And I cannot find it is convert to URCommander.

Thank you.

gavanderhoorn, 2019-04-30:
Are you specifically asking about the `LowBandwidthTrajectoryFollower`?

gavanderhoorn, 2019-04-30:
As this is not an actual issue with the package in this repository, but a general question about an implementation detail, I'm going to close this issue.

You may keep commenting on it of course.

closed 2019-04-30 by gavanderhoorn

itfanr, 2019-04-30:
@gavanderhoorn Thank you all the same.

 If you have known about this, could you tell me the implementation detail ?

gavanderhoorn, 2019-04-30:
You haven't answered my question:

> Are you specifically asking about the `LowBandwidthTrajectoryFollower`?

itfanr, 2019-04-30:
@gavanderhoorn  I want know  the function how does `bool LowBandwidthTrajectoryFollower::execute`  send the command to real UR robot with  30001 port. I didnot find the code.

itfanr, 2019-04-30:
@gavanderhoorn It must call URCommander or some code. But I cannot figure it out.