ros-industrial-attic/ur_modern_driver #304: error in the README.md

cocodmdr, 2019-05-01:
Lost quite a lot of time following the instructions in the readme.md of branch master for a stupid error ...

In : Installation > Usage 
 
roslaunch urXX_moveit_config ur5_moveit_planning_executing.launch

It shoud be instead roslaunch urXX_moveit_config ur5_moveit_planning_executi**on**.launch

gavanderhoorn, 2019-05-01:
> Lost quite a lot of time following the instructions in the readme.md of branch master [..]

This is a duplicate of #198.

That issue is closed as it is fixed in `kinetic-devel`.

As `master` is deprecated (notice at the top of the file) and will not see any development we'll not fix it there.

gavanderhoorn, 2019-05-01:
Closing as it's a duplicate.

gavanderhoorn, 2019-05-01:
I sympathise that you lost time with this of course.

Not dismissing that, but we have to make choices where we spent our (limited) time.

closed 2019-05-01 by gavanderhoorn