ros-industrial-attic/ur_modern_driver #307: velocity control on UR5 (kinetic)

Gloriabhsfer, 2019-05-18:
Hi,
I want to control a real UR5 robot to do a simple pick and place repeat movement. I choose to use ur_modern_driver to communicate with the robot. I firstly try to use the testmove.py file. It works, but when the program is started, it takes about 5 mins before the robot move( which is a relatively long time if I need to to real-time communication in future). In my understanding, the velocities in testmove is the derivate of distance between the waypoints I set. And if the speed is over the max speed, the robot will stop even if it's not the singularity position.But for safety consideration, I cannot let the max speed of UR5 too large and have the risk to damage it self. So I think a better way for me is to do velocity control of UR5. 
I have searched on the issue of ur_modern_driver, most of the people will publish some velocity on specific joint to control it through /ur_driver/joint_state. But when I try this way, the ros node have information I published but robot will not move. 
```
import rospy
import actionlib
from sensor_msgs.msg import JointState
from std_msgs.msg import Header
from control_msgs.msg import *
from trajectory_msgs.msg import *

JOINT_NAMES = ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint',
               'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']

#global k
#to talk to virtual model
def talker():
	global joint_pos
	global jvel1
	global jvel2
	global jvel
	theta=0
	pub = rospy.Publisher('/ur_driver/joint_speed', JointTrajectory, queue_size=10)
	rospy.init_node('pttraj')
	rate = rospy.Rate(125) # 10hz
	hello_str = JointTrajectory()
	hello_str.header = Header()
	hello_str.joint_names=JOINT_NAMES
	joint_states=rospy.wait_for_message("joint_states", JointState)
	jvel=joint_states.position[0]
	while not rospy.is_shutdown():
		joint_states=rospy.wait_for_message("joint_states", JointState)
		jvel1=joint_states.position[0]
		jvel2=jvel1-jvel
		jvel2=jvel2*180/3.14
		hello_str.points=[
				JointTrajectoryPoint(velocities=(-1,0,0,0,0,0), time_from_start=rospy.Duration(0.0)),
				JointTrajectoryPoint(time_from_start=rospy.Duration(0.0))]
		pub.publish(hello_str)
		rate.sleep()
		#print k
		hello_str.header.seq=hello_str.header.seq+1
		hello_str.header.stamp = rospy.Time.now()
		pub.publish(hello_str)
		rate.sleep()
if __name__ == '__main__':
	try:
		talker()
	except rospy.ROSInterruptException:
		pass
```
And I find many people use "speedj", it that form ros_control?URScript?or ur_modern_driver? I don't know how to implement the control structure in it.
For future research consideration, I think a better way for me is to find a better way to do the velocity control of UR5. Hope anyone can give me some idea or suggestion to do that.
Best,
Gloria

gavanderhoorn, 2019-05-20:
Could you please first clarify what it is you actually want to do?

The title of your issue mentions "velocity control", but the body then talks about pick-and-place, `testmove.py` and `/ur_driver/joint_state`. That doesn't make much sense to me, so it'd be good if you can summarise your goal for us.

Velocity control is certainly possible, but not by using any of the approaches you've shown or referred to.

Please also tell us whether you're using `master` or `kinetic-devel` (as the deprecation notice on `master` states you should use).

Gloriabhsfer, 2019-05-21:
Thanks for your reply. I want to control the UR5 in Velocity and acceleration control. And I want to start with a simple task like pick and place. 
I use the `kinetic-level` package with ubuntu16.04. It seems in testmove.py it's only a position base control and our UR5 will have some weird performance. Sometimes it will over the max speed so the robot will stop. So I think a better and more safe way is to do the velocity and acceleration control.
When I sent the joint speed to joint_state massage, the rostopic will show the change to each joint speed, but robot won't move.
Can you gave me some idea to do the velocity control?

gavanderhoorn, 2019-05-22:
> 
> 
> Thanks for your reply. I want to control the UR5 in Velocity and acceleration control. And I want to start with a simple task like pick and place.

re: acceleration control: if you mean force/torque/effort control: that is not supported by `ur_modern_driver`.

> I use the `kinetic-level` package with ubuntu16.04. It seems in testmove.py it's only a position base control and our UR5 will have some weird performance. Sometimes it will over the max speed so [...]

The most straightforward way is probably using the `ros_control` mode of the driver, activating the `joint_group_vel_controller` controller and then sending velocity setpoints to it over the exposed topic.

See #218 for some related discussion.

gavanderhoorn, 2019-05-22:
As this is not actually an issue with the package in this repository, I'm going to close it.

Feel free to keep commenting of course @Gloriabhsfer.

closed 2019-05-22 by gavanderhoorn