ros-industrial-attic/ur_modern_driver #311: Socket

KushaagraGoyal, 2019-06-03:


gavanderhoorn, 2019-06-03:
I'm confused.

Can you give a bit more detail as to what problem you are encountering?

KushaagraGoyal, 2019-06-03:
Hi,

First of all, I am really sorry for the above confusion, its my first time using github.

I am using UR5e CB2 with controller URControl 1.8.25319, which I am trying to connect to my ROS installed in Ubuntu 16.04 on Oracle VM VirtualBox, through a router. Upon executing the three launching commands, first to bring up the ur, second to load all the  nodes for moveit and the third to launch the Rviz simulation, the code runs, no error are shown there. Then, I set the final state in rviz simuation, and clicked on the Plan and execute button. There aren't any problems shown in running these commands on the terminal. However, the error (shown in the image below) is popping up on my pendent's run logs as soon as I press the button.

[image]

At first I thought this was just the networking error but the rviz simulation is showing the correct joint states of the bot. Hence, there shouldn't be any error in the connection. 
Its just that the driver is not able to evoke the joint controllers (my speculation).

Please help me through it!
Thanks.

gavanderhoorn, 2019-06-03:
> I am using UR5e CB2 with controller URControl 1.8.25319,

Is that really a UR5**e**? Or a regular UR5?

>  which I am trying to connect to my ROS installed in Ubuntu 16.04 on Oracle VM VirtualBox, through a router.

This is the cause of your problems.

If with "through a router" you mean: my VM has a NAT connection to my PC, then you need to configure the VM to allow connections on port 50001 to reach the guest OS.

if with "through a router" you mean: I have a physical router between me and the robot, then if the router is a NAT device (and is actually NAT-ting between your PC and the robot), you'll need to configure the router to allow connections from the robot through the NAT to your PC.

gavanderhoorn, 2019-06-03:
Note that with #301 merged we have better support for NAT-ed setups (as you can configure a different IP for the return connection).

gavanderhoorn, 2019-06-03:
As this is a local configuration issue and most likely not an issue with the package in this repository, I'm closing this.

Feel free to keep commenting on it of course @KushaagraGoyal.

closed 2019-06-03 by gavanderhoorn