ros-industrial-attic/ur_modern_driver #312: real ur5 does not move using moveit_planning_execution.launch

IndustrialEngStudent, 2019-06-05:
Hi,

I am trying to move the ur5 + Robotiq gripper in a small circle.
First I bring up the ur5 + robotiq gripper:

```
 roslaunch ur_modern_driver ur5_robotiq_2f85_bringup2.launch robot_ip:=192.168.1.151  [reverse_port:=REVERSE_PORT]
... logging to /home/laurah/.ros/log/2b623334-878e-11e9-92dc-8086f2694d08/roslaunch-laurah-HP-EliteBook-840-G1-17016.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laurah-HP-EliteBook-840-G1:44949/

SUMMARY
========

PARAMETERS
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /gripper/hardware_control_loop/loop_hz: 125
 * /gripper/hardware_interface/joints: ['gripper_finger1...
 * /gripper/joint_position_controller/action_monitor_rate: 10
 * /gripper/joint_position_controller/constraints/goal_time: 0.6
 * /gripper/joint_position_controller/constraints/simple_gripper_robotiq_85_left_knuckle_joint/goal: 0.01
 * /gripper/joint_position_controller/constraints/simple_gripper_robotiq_85_left_knuckle_joint/trajectory: 0.1
 * /gripper/joint_position_controller/constraints/stopped_velocity_tolerance: 0.05
 * /gripper/joint_position_controller/joint: gripper_finger1_j...
 * /gripper/joint_position_controller/state_publish_rate: 125
 * /gripper/joint_position_controller/stop_trajectory_duration: 0.5
 * /gripper/joint_position_controller/type: position_controll...
 * /gripper/joint_state_controller/publish_rate: 125
 * /gripper/joint_state_controller/type: joint_state_contr...
 * /gripper/joint_trajectory_controller/joint: ['gripper_finger1...
 * /gripper/joint_trajectory_controller/type: position_controll...
 * /hardware_control_loop/loop_hz: 125
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_based_pos_traj_controller/action_monitor_rate: 10
 * /pos_based_pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/elbow_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/constraints/goal_time: 0.6
 * /pos_based_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_based_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_based_pos_traj_controller/state_publish_rate: 125
 * /pos_based_pos_traj_controller/stop_trajectory_duration: 0.5
 * /pos_based_pos_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robotiq_85_driver/baud: 115200
 * /robotiq_85_driver/comport: /dev/ttyUSB0
 * /robotiq_85_driver/num_grippers: 1
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /ur_driver/base_frame: base
 * /ur_driver/max_joint_difference: 0.01
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
[... 94 lines cut ...]
[ INFO] [1559742970.218482071]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1559742970.238703477]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1559742970.238764606]: Starting main loop
[ INFO] [1559742970.238990497]: Setting up connection: 192.168.1.151:30003
[ INFO] [1559742970.239082006]: Starting pipeline RTPacket
[ INFO] [1559742970.239216833]: Setting up connection: 192.168.1.151:30002
[ INFO] [1559742970.239299913]: Starting pipeline StatePacket
[ INFO] [1559742970.240570777]: Connection established for 192.168.1.151:30003
[ INFO] [1559742970.240864038]: Connection established for 192.168.1.151:30002
[ INFO] [1559742970.242381435]: Starting ActionServer
[ INFO] [1559742970.242544963]: Trajectory thread started
```
Then: `roslaunch lau_moveit_config ur5_robotiq_2f85_moveit_planning_execution.launch`
which give me no trouble execpt two warnings:
[ WARN] [1559742315.209671241]: Skipping virtual joint 'virtual_joint' because its child frame 'base_link' does not match the URDF frame 'world'
...
`[ WARN] [1559742489.710331295]: The complete state of the robot is not yet known.  Missing gripper_finger1_joint`

Then: `roslaunch lau_moveit_config moveit_rviz.launch config:=true`, which is also not a problem.

Then, I run the script using the move group python interface: " rosrun ur5_notebook motionplanner1LauCircle_gripper.py"

In rviz I see the robot moving as desired. The starting position is the exact position as the real robot is. Thus, I think the connection is working since rviz shows the real state of the robot. However, the real ur5 does not move. 

How can I fix this?

miguelprada, 2019-06-05:
It's impossible for us to help you with this, since we have no idea of what are the contents of `ur5_robotiq_2f85_bringup2.launch` or anything in `lau_moveit_config` or `ur5_notebook` packages. You need to be much more specific to ask for help.

Since this doesn't seem to be a problem related to this driver, but rather to your application configuration, I will close this issue. Feel free to keep commenting or ask in [ROS Answers](https://answers.ros.org/questions/) if you have general questions about configuring a ROS-based application.

closed 2019-06-05 by miguelprada

gavanderhoorn, 2019-06-05:
I would agree with @miguelprada that this issue seems a bit too localised for us to address here: we don't know how you configured things, nor does `ur_modern_driver` have anything to do with robotiq grippers.

However, if you can reproduce your problem with *just* the files from this repository, then we can try to see whether we can resolve it.

nag92, 2019-07-08:
I am have the same issue with just the vanilla ur_modern_driver and ur_robot packages

gavanderhoorn, 2019-07-08:
Please do not comment on closed issues.

Open a new one. If it's really the same issue, we'll close the new one as a duplicate and re-open this one.