ros-industrial-attic/ur_modern_driver #313: UR10e stops mid path

ThomHordijk, 2019-06-07:
Recently we ran into an issue controlling our UR10e. We are using Ubuntu 16.04, ROS kinetic,  Moveit, descartes, this package and the modern driver. The cobot can execute its path a number of time, or just once and then it will stop mid path. It does not give an error. The next time the ur driver tries to establish a connection it states that it cannot. We have tried to look into the error and have seen similair behavior when the speed of the cobot was set to low, this caused timing miss matches, however, increasing it did not solve the problem. We have not found similair issues online so we hope you can help resolve the problem.

miguelprada, 2019-06-07:
I transferred the issue, since it seems more related to the driver itself.

I guess you're using the version in #216, right?

In which mode are you starting the driver? Can you check which error code is returned in the `FollowJointTrajectoryAction` result message?

ThomHordijk, 2019-06-14:
Hi, we are using the kinetic-devel version of the driver. I am not sure if this is the error you are aiming for but what i can find in the logs is :

` 1560495297.680455572 ERROR [/home/riwo-rack-pc/GIT/automatic-planthandling/Ros/catkin_ws/src/library/ur_modern_driver/src/ros/lowbandwidth_trajectory_follower.cpp:302(start) [topics: /rosout, /joint_states, /wrench, /tool_velocity, /joint_temperature, /tf, /ur_driver/io_states, /ur_driver/robot_status, /ur_driver/masterboard_state, /ur_driver/robot_mode_state, /follow_joint_trajectory/result, /follow_joint_trajectory/feedback, /follow_joint_trajectory/status] Failed to accept incoming robot connection
`
The question about the mode in which we start the driver is somewhat unclear to me, could you elaborate?

Drikuy, 2019-07-08:
Hi, I have the same problem and in my UR10e robot. The thing is that when I use Remote Control Mode and I have a trajectory planned it stops with the following message on the teaching pendant:
> C171A6: Issue with blends: Overlapping Blends in a MoveJ, a waypoint was skipped
One message of this kind for each skipped waypoint.
And the message in the terminal is the following one:
`ABORTED: Solution found but controller failed during execution`

gavanderhoorn, 2019-11-23:
Please use `ur_robot_driver` for all CB3 and e-series controllers.

If the issue reported in the OP is also observed with that package, please report it on the `ur_robot_driver` issue tracker.

closed 2019-11-23 by gavanderhoorn