ros-industrial-attic/ur_modern_driver #316: Sub-package of type 3 was not parsed completely

sashaiw, 2019-06-24:
I am trying to control a UR5 CB3 running Polyscope 3.10 using `ur_modern_driver`. When I launch `ur5_bringup.launch` I get the following message:

`Sub-package of type 3 was not parsed completely!`

I launch it with `roslaunch ur_modern_driver ur5_bringup.launch robot_ip=ROBOT_IP`. I am running the `kinetic-devel` branch for both `ur_modern_driver` and `universal_robot`.

A complete log follows:
```
$ roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.3
... logging to /home/csrobot/.ros/log/658d005c-96c0-11e9-8102-04d4c40355e0/roslaunch-scooter-7363.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.2:37579/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /ur_driver/base_frame: base
 * /ur_driver/max_joint_difference: 0.01
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/max_waiting_time: 2.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/require_activation: Never
 * /ur_driver/reverse_ip_address: 
 * /ur_driver/reverse_port: 50001
 * /ur_driver/robot_ip_address: 192.168.1.3
 * /ur_driver/servoj_gain: 100.0
 * /ur_driver/servoj_lookahead_time: 1.0
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/servoj_time_waiting: 0.001
 * /ur_driver/shutdown_on_disconnect: True
 * /ur_driver/time_interval: 0.008
 * /ur_driver/tool_frame: tool0_controller
 * /ur_driver/use_lowbandwidth_trajectory_follower: False
 * /ur_driver/use_ros_control: False

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [7378]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 658d005c-96c0-11e9-8102-04d4c40355e0
process[rosout-1]: started with pid [7391]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [7408]
process[ur_driver-3]: started with pid [7409]
[ INFO] [1561408869.898002830]: Setting up connection: 192.168.1.3:30003
[ INFO] [1561408869.898329428]: Connection established for 192.168.1.3:30003
[ INFO] [1561408869.898345361]: Setting up connection: 192.168.1.3:30001
[ INFO] [1561408869.898662941]: Connection established for 192.168.1.3:30001
[ INFO] [1561408869.952301031]: Got VersionMessage:
[ INFO] [1561408869.952323285]: project name: URControl
[ INFO] [1561408869.952332086]: version: 3.10.0
[ INFO] [1561408869.952340036]: build date: 03-05-2019, 06:39:00
[ INFO] [1561408869.952347934]: Disconnecting from 192.168.1.3:30001
[ INFO] [1561408869.958821522]: ActionServer enabled
[ INFO] [1561408869.958851287]: Use standard trajectory follower
[ INFO] [1561408869.960502838]: Setting up connection: :50001
[ INFO] [1561408869.960606243]: Connection established for :50001
[ INFO] [1561408869.960663930]: Initializing ur_driver/URScript subscriber
[ INFO] [1561408869.964247017]: The ur_driver/URScript initialized
[ INFO] [1561408869.964273637]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1561408869.969692244]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1561408869.969725988]: Starting main loop
[ INFO] [1561408869.969824551]: Setting up connection: 192.168.1.3:30003
[ INFO] [1561408869.969904525]: Starting pipeline RTPacket
[ INFO] [1561408869.970000586]: Starting pipeline StatePacket
[ INFO] [1561408869.970073323]: Setting up connection: 192.168.1.3:30002
[ INFO] [1561408869.970199745]: Connection established for 192.168.1.3:30003
[ INFO] [1561408869.970419376]: Connection established for 192.168.1.3:30002
[ INFO] [1561408869.971958574]: Starting ActionServer
[ INFO] [1561408869.972097937]: Trajectory thread started
[ERROR] [1561408870.028391551]: Sub-package of type 3 was not parsed completely!
[ INFO] [1561408870.028497411]: Disconnecting from 192.168.1.3:30002
[ INFO] [1561408870.028592966]: Shutting down on stopped pipeline StatePacket
*** Error in `/home/csrobot/catkin_ws/devel/lib/ur_modern_driver/ur_driver': corrupted double-linked list: 0x00000000012788c0 ***
[ur_driver-3] process has died [pid 7409, exit code -11, cmd /home/csrobot/catkin_ws/devel/lib/ur_modern_driver/ur_driver __name:=ur_driver __log:=/home/csrobot/.ros/log/658d005c-96c0-11e9-8102-04d4c40355e0/ur_driver-3.log].
log file: /home/csrobot/.ros/log/658d005c-96c0-11e9-8102-04d4c40355e0/ur_driver-3*.log
```

azhural, 2019-06-26:
I just upgraded to 3.10 and am experiencing the same issue. Looking at the 3.10 Release notes my guess would be, that it has something to do with

>     Added reserved byte to Masterboard data sub package of Robot State Message.

sashaiw, 2019-07-01:
I can confirm that downgrading to 3.9.1 worked for me.

Abduoit, 2019-11-19:
Did you solve the issue, can we have `UR5 CB3 software 3.10` working with `ur_modern_driver` ?

gavanderhoorn, 2019-11-19:
@Abduoit: use [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) for all CB3 and e-series robot controllers. See the readme there for additional requirements and setup instructions.

gavanderhoorn, 2019-11-23:
I'm closing this as we've officially deprecated this package.

Refer to the [announcement on ROS Discourse](https://discourse.ros.org/t/deprecation-of-ur-modern-driver/11629).

closed 2019-11-23 by gavanderhoorn

jc-sanchez16, 2021-09-09:
Having this error again on SW version  3.14

gavanderhoorn, 2021-09-10:
Please see the last few comments on this issue.

You should not be using `ur_modern_driver` with a CB3.

Please use [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver).