ros-industrial-attic/ur_modern_driver #317: UR10e robot connection but not moving

Drikuy, 2019-06-26:
Hi,
I am new in ROS and I am having some troubles to move UR10e robot (real) using ROS. The thing is that I connect with the robot, but when I use MoveIt to load the planned path it just stops. To launch the robot communication and its controller I use the following statements:

`roslaunch ur_modern_driver ur10e_bringup.launch robot_ip:=192.168.0.100`
`roslaunch ur10_e_moveit_config ur10_e_moveit_planning_execution.launch `
`roslaunch ur10_e_moveit_config moveit_rviz.launch`

Here is the code of the terminal window where I have launched ur10e_bringup.launch file:
`$ roslaunch ur_modern_driver ur10e_bringup.launch robot_ip:=192.168.0.100
... logging to /home/diego/.ros/log/d2a8e5a8-97f5-11e9-8c12-28f10e0c925d/roslaunch-ubuntu-10635.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:39197/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /ur_driver/base_frame: base
 * /ur_driver/max_joint_difference: 0.01
 * /ur_driver/max_payload: 10.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/max_waiting_time: 2.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/require_activation: Never
 * /ur_driver/reverse_ip_address: 
 * /ur_driver/reverse_port: 50001
 * /ur_driver/robot_ip_address: 192.168.0.100
 * /ur_driver/servoj_gain: 100.0
 * /ur_driver/servoj_lookahead_time: 1.0
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/servoj_time_waiting: 0.001
 * /ur_driver/shutdown_on_disconnect: True
 * /ur_driver/time_interval: 0.008
 * /ur_driver/tool_frame: tool0_controller
 * /ur_driver/use_lowbandwidth_trajectory_follower: False
 * /ur_driver/use_ros_control: False

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [10648]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d2a8e5a8-97f5-11e9-8c12-28f10e0c925d
process[rosout-1]: started with pid [10661]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [10678]
process[ur_driver-3]: started with pid [10679]
[ INFO] [1561541767.973353797]: Setting up connection: 192.168.0.100:30003
[ INFO] [1561541767.973751898]: Connection established for 192.168.0.100:30003
[ INFO] [1561541767.973786777]: Setting up connection: 192.168.0.100:30001
[ INFO] [1561541767.974005093]: Connection established for 192.168.0.100:30001
[ INFO] [1561541768.376076977]: Got VersionMessage:
[ INFO] [1561541768.376181783]: project name: URControl
[ INFO] [1561541768.376217428]: version: 5.1.2
[ INFO] [1561541768.376278614]: build date: 05-10-2018, 13:34:19
[ INFO] [1561541768.376332309]: Disconnecting from 192.168.0.100:30001
[ INFO] [1561541768.401965269]: ActionServer enabled
[ INFO] [1561541768.402074225]: Use standard trajectory follower
[ INFO] [1561541768.408230833]: Setting up connection: :50001
[ INFO] [1561541768.408696116]: Connection established for :50001
[ INFO] [1561541768.408992310]: Initializing ur_driver/URScript subscriber
[ INFO] [1561541768.424832385]: The ur_driver/URScript initialized
[ INFO] [1561541768.424921719]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1561541768.439805581]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1561541768.439879972]: Starting main loop
[ INFO] [1561541768.440084619]: Setting up connection: 192.168.0.100:30003
[ INFO] [1561541768.440230648]: Starting pipeline RTPacket
[ INFO] [1561541768.440560871]: Setting up connection: 192.168.0.100:30002
[ INFO] [1561541768.440671680]: Connection established for 192.168.0.100:30003
[ INFO] [1561541768.440778690]: Starting pipeline StatePacket
[ INFO] [1561541768.441130794]: Connection established for 192.168.0.100:30002
[ INFO] [1561541768.444258020]: Starting ActionServer
[ INFO] [1561541768.444421148]: Trajectory thread started
[ INFO] [1561541810.803630215]: Received new goal
[ INFO] [1561541810.803849968]: Trajectory received and accepted
[ INFO] [1561541810.803965522]: Translating trajectory
[ INFO] [1561541810.804049153]: Executing trajectory with 31 points and duration of 40.000s
[ INFO] [1561541810.804125314]: Attempting to start follower 0x1830d50
[ INFO] [1561541810.804281190]: Uploading trajectory program to robot`

In addition, I left here an *.jpg picture of the rqt_graph when the system is working.
(https://user-images.githubusercontent.com/33570252/60170729-5b930480-9809-11e9-839e-06136737c3b5.jpg)![rqt_graph]

May the problem be that execute_trajectory must be linked with follow_joint_trajectory? If so, I think that the solution is remapping one of this two topics to the other. Does it make any sense? 
Thank for your help

martimorta-wood, 2019-06-27:
Hi @Drikuy, 
I used the driver last week and it was working, anyway, have you:
- Selected "Remote Control" in the teaching pendant?
- Selected very low speed and acceleration scale values in MoveIt (I think I am using something around 0.05)
Cheers

Drikuy, 2019-06-27:
Hi @martimorta-wood , thank you for answering so quickly.

- I have the "Remote Control" of the teaching pendant enabled. Do I have to switch on any other config in order to enable the "Remote Control"?
- I try with very low speed and acceleration in MoveIt (values between 0.01 and 0.05) and it still does not move. The terminal 1 and 2 (where I am running ur_modern drivers / ur10e_bringup.launch and ur10_e_moveit_planning_execution.launch) shows the following messages:

Terminal 1:

> [ INFO] [1561636317.776657282]: Setting up connection: 192.168.1.100:30003
[ INFO] [1561636317.776869256]: Connection established for 192.168.1.100:30003
[ INFO] [1561636317.776900328]: Setting up connection: 192.168.1.100:30001
[ INFO] [1561636317.777089206]: Connection established for 192.168.1.100:30001
[ INFO] [1561636317.880852874]: Got VersionMessage:
[ INFO] [1561636317.880930517]: project name: URControl
[ INFO] [1561636317.880977149]: version: 5.1.2
[ INFO] [1561636317.881016174]: build date: 05-10-2018, 13:34:19
[ INFO] [1561636317.881057977]: Disconnecting from 192.168.1.100:30001
[ INFO] [1561636317.906152235]: ActionServer enabled
[ INFO] [1561636317.906282383]: Use standard trajectory follower
[ INFO] [1561636317.912644297]: Setting up connection: :50001
[ INFO] [1561636317.913055656]: Connection established for :50001
[ INFO] [1561636317.913269978]: Initializing ur_driver/URScript subscriber
[ INFO] [1561636317.929498077]: The ur_driver/URScript initialized
[ INFO] [1561636317.929597071]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1561636317.943178338]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1561636317.943266552]: Starting main loop
[ INFO] [1561636317.943477902]: Setting up connection: 192.168.1.100:30003
[ INFO] [1561636317.943626495]: Starting pipeline RTPacket
[ INFO] [1561636317.943791015]: Starting pipeline StatePacket
[ INFO] [1561636317.943899414]: Connection established for 192.168.1.100:30003
[ INFO] [1561636317.944473180]: Setting up connection: 192.168.1.100:30002
[ INFO] [1561636317.944736659]: Connection established for 192.168.1.100:30002
[ INFO] [1561636317.946713646]: Starting ActionServer
[ INFO] [1561636317.946889595]: Trajectory thread started
[ INFO] [1561636388.689623224]: Received new goal
[ INFO] [1561636388.689703620]: Trajectory received and accepted
[ INFO] [1561636388.689779118]: Translating trajectory
[ INFO] [1561636388.689822391]: Executing trajectory with 10 points and duration of 10.223s
[ INFO] [1561636388.689878011]: Attempting to start follower 0x1414d50
[ INFO] [1561636388.689910608]: Uploading trajectory program to robot

Terminal 2:
You can start planning now!

[ INFO] [1561636367.062867049]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1561636367.063547080]: Goal constraints are already satisfied. No need to plan or execute any motions
[ INFO] [1561636388.611938894]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1561636388.612449727]: Planning attempt 1 of at most 1
[ INFO] [1561636388.618726923]: Planner configuration 'manipulator[RRTConnectkConfigDefault]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1561636388.622419273]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1561636388.623150911]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1561636388.623455971]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1561636388.626962569]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1561636388.640522777]: manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[ INFO] [1561636388.641476500]: manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[ INFO] [1561636388.644016349]: manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[ INFO] [1561636388.652035037]: manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[ INFO] [1561636388.652744876]: ParallelPlan::solve(): Solution found by one or more threads in 0.031634 seconds
[ INFO] [1561636388.656066876]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1561636388.656557745]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1561636388.660477094]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1561636388.663543636]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1561636388.664960505]: manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[ INFO] [1561636388.668973895]: manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[ INFO] [1561636388.670326206]: manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[ INFO] [1561636388.671744321]: manipulator[RRTConnectkConfigDefault]: Created 5 states (3 start + 2 goal)
[ INFO] [1561636388.672047506]: ParallelPlan::solve(): Solution found by one or more threads in 0.016969 seconds
[... cut ...]
[ INFO] [1561636388.675537696]: manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[ INFO] [1561636388.675850521]: ParallelPlan::solve(): Solution found by one or more threads in 0.003513 seconds
[ INFO] [1561636388.686769315]: SimpleSetup: Path simplification took 0.010675 seconds and changed from 3 to 2 states

After those messages the program seems to stop. I think that the connection with the robot is done, but I don't know why it cannot moves. Do I need to make any other special configuration from the teach pendant to allow Remote Control with the robot?

martimorta-wood, 2019-06-27:
> * I have the "Remote Control" of the teaching pendant enabled. Do I have to switch on any other config in order to enable the "Remote Control"?

No

> I think that the connection with the robot is done, but I don't know why it cannot moves. Do I need to make any other special configuration from the teach pendant to allow Remote Control with the robot?

You can check if the connection is done if in RViz the arm is shown in its current position or if you move it around with the teach pendant changes the model in RViz.
About the Remote control, just be sure you do it before starting the whole ROS nodes.

Drikuy, 2019-06-28:
I've started properly the Remote Control, but it still does not work. The moveit launch file I'm using is the following one:
`roslaunch ur10_e_moveit_config ur10_e_moveit_planning_execution.launch`
Should I use this other `roslaunch ur_modern_driver ur10e_ros_control.launch robot_ip:=ROBOT_IP` one to launch ROS real controllers? 
When I try to launch `ur10e_ros_control.launch` it stops the `/ur_driver`, so that is why I am not using that second launch file.

Drikuy, 2019-06-28:
I've been trying with `roslaunch ur_modern_driver ur10e_ros_control.launch robot_ip:=10.42.0.100` instead of using `roslaunch ur_modern_driver ur10e_bringup.launch robot_ip:=10.42.0.100` to connect with UR10e robot and it works too. the thing is that when I start the two followings launch files:
`roslaunch ur10_e_moveit_config ur10_e_moveit_planning_execution.launch limited:=true`
`roslaunch ur10_e_moveit_config moveit_rviz.launch`
After clicking in `Plan and Execute` RViz button I get the following error code:

> [ INFO] [1561712544.036737759]: Received new trajectory execution service request...
[ WARN] [1561712544.345255860]: Controller  failed with error code PATH_TOLERANCE_VIOLATED
[ WARN] [1561712544.345666815]: Controller handle  reports status ABORTED
[ INFO] [1561712544.345875103]: Completed trajectory execution with status ABORTED ...
[ INFO] [1561712544.346312584]: Execution completed: ABORTED

Does anyone knows how to solve this?

abdelhbibbounou, 2019-06-28:
hello,
Me too i have the same problem, moreover i launched the ur5e_ros_control launch files and i tried to publish a message into the "  /pos_based_pos_traj_controller/command" topic but i got the following error : [ERROR] [1561713898.932264704]: Can't accept new commands. Controller is not running.

mlautman, 2019-06-29:
+1 however I am running the UR5 with polyscope version 3.5.3

gavanderhoorn, 2019-06-29:
> +1 however I am running the UR5 with polyscope version 3.5.3

So that would make your comment off-topic, as it's a different robot, different controller and a different version of Polyscope.

Drikuy, 2019-07-02:
Th UR10e robot still does not move, but I think I have improve a little. Using ROS control launch file for UR10e controllers I've made the connection between ROS and the robot because I can "echo" subscribe to robot topics as `/ur_driver/robot_status`. I also have connection between `/move_group` node and the `hardware_interface` node as you can see in the following `rqt_graph` window:
[image]
But when I launch the following launch files:
`roslaunch ur_modern_driver ur10e_ros_control.launch robot_ip:=10.42.0.100` (Terminal 1)
`roslaunch ur10_e_moveit_config ur10_e_moveit_planning_execution.launch limited:=true ` (Terminal 2)
`roslaunch ur10_e_moveit_config moveit_rviz.launch` (Terminal 3)
I've get the following terminal responses:
- Terminal 1

> ...
> [INFO] [1562046728.950168]: Started controllers: joint_state_controller, force_torque_sensor_controller, vel_based_pos_traj_controller
> [ros_control_controller_manager-4] process has finished cleanly
> log file: /home/diego/.ros/log/864ee6f2-9c8d-11e9-a1b8-000c299380fc/ros_control_controller_manager-4*.log
> [ERROR] [1562046801.857816175]: Can't accept new action goals. Controller is not running.

- Terminal 2

> ...
> [ INFO] [1562046801.845601251]: manipulator[RRTConnectkConfigDefault]: Created 5 states (2 start + 3 goal)
> [ INFO] [1562046801.846909472]: ParallelPlan::solve(): Solution found by one or more threads in 0.002234 seconds
> [ INFO] [1562046801.850152676]: SimpleSetup: Path simplification took 0.002486 seconds and changed from 3 to 2 states
> [ERROR] [1562046801.858481437]: Trying to getState() when no goal is running. You are incorrectly using SimpleActionClient
> [ WARN] [1562046801.858568597]: Controller pos_based_pos_traj_controller failed with error code INVALID_GOAL
> [ WARN] [1562046801.858845545]: Controller handle pos_based_pos_traj_controller reports status RUNNING

- Terminal 3

> [ INFO] [1562046733.637738147]: rviz version 1.12.17
> [ INFO] [1562046733.637815924]: compiled against Qt version 5.5.1
> [ INFO] [1562046733.637843212]: compiled against OGRE version 1.9.0 (Ghadamon)
> [ INFO] [1562046733.873256504]: Stereo is NOT SUPPORTED
> [ INFO] [1562046733.873423152]: OpenGl version: 2.1 (GLSL 1.2).
> [ INFO] [1562046737.316844538]: Loading robot model 'ur10e'...
> [ WARN] [1562046737.317006222]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
> [ INFO] [1562046737.317069524]: No root/virtual joint specified in SRDF. Assuming fixed joint
> [ INFO] [1562046737.504082000]: Loading robot model 'ur10e'...
> [ WARN] [1562046737.504376559]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
> [ INFO] [1562046737.506813041]: No root/virtual joint specified in SRDF. Assuming fixed joint
> [ INFO] [1562046737.658855502]: Starting scene monitor
> [ INFO] [1562046737.667726188]: Listening to '/move_group/monitored_planning_scene'
> [ INFO] [1562046738.362278992]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
> [ INFO] [1562046739.456152844]: Ready to take commands for planning group manipulator.
> [ INFO] [1562046739.456253783]: Looking around: no
> [ INFO] [1562046739.456276805]: Replanning: no
> [ WARN] [1562046739.474013636]: Interactive marker 'EE:goal_ee_link' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.
> [ INFO] [1562046802.055633486]: ABORTED: Solution found but controller failed during execution

So I really don't know what I'm doing wrong, because the controllers seems to be launched and connected but when I try to use them it just doesn't respond.

Drikuy, 2019-07-02:
[Solved]
I was making a wrong configuration of the _Remote Control_ mode of the robot. I was able to move the robot using RViz and these three launch files:
```
$ roslaunch ur_modern_driver ur10e_bringup.launch robot_ip:=10.42.0.100
$ roslaunch ur10_e_moveit_config ur10_e_moveit_planning_execution.launch 
$ roslaunch ur10_e_moveit_config moveit_rviz.launch
```
Thank you everyone so much for your help!

closed 2019-07-02 by Drikuy

gavanderhoorn, 2019-07-02:
Good to hear you got it to work in the end.

Remote control is definitely something that needs to be correctly setup, otherwise things won't work.

The new driver (FZI&UR) will allow for a different way of enabling the driver to control the robot which should make this much less of a hassle.

skywoodsz, 2019-11-04:
> [Solved]
> I was making a wrong configuration of the _Remote Control_ mode of the robot. I was able to move the robot using RViz and these three launch files:
>
> ```
> $ roslaunch ur_modern_driver [...]
excuse me ,I'm sorry to disturb you，i would like to know how to solve the problems,please,thank you very much.

Drikuy, 2019-11-08:
I'm sorry @skywoodsz for answering so late, but I've been sick the whole week and I have no strenght to get myself in front of the computer. ROS files that are given in this repository works very well. The problem is that I did not set up propperly the robot on the teaching pendant.

To do so, you must enable remote control on the teaching pendant settings (you can access this menu by the hamburguer menu at the top-right corner of the teaching pendant). Very close to that hamburguer menu, there must be an icon with the shape af a teaching pendant, click on it and it will allow you to put the robot on remote control. 

If you cannot make this work feel free to ask as many times as you need :)

angux94, 2021-06-07:
Hello! I'm facing the same problem as the one in this post, but my robot is a UR5 series, I can't seem to find the "enable remote control" button or settings, but for the most part I have the same issue, I can plan and the trajectory remains uploading to the robot but it never moves.

Any idea on where can I enable remote control in the teaching pendant?

In my case:

- Terminal 1: moveit_planning_execution
`[ INFO] [1623080452.620766800]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1623080452.624849400]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623080452.626060700]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623080452.627358400]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623080452.630102600]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623080452.632353900]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623080452.637992300]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1623080452.640879100]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623080452.641072900]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623080452.641454900]: ParallelPlan::solve(): Solution found by one or more threads in 0.018031 seconds
[ INFO] [1623080452.642295300]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623080452.642572900]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623080452.644088400]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623080452.645325400]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623080452.646591100]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623080452.647176200]: RRTConnect: Created 5 states (3 start + 2 goal)
[ INFO] [1623080452.648139100]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623080452.649973400]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623080452.650326900]: ParallelPlan::solve(): Solution found by one or more threads in 0.008326 seconds
[ INFO] [1623080452.650882200]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623080452.656029400]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623080452.657866500]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623080452.660899000]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623080452.661241900]: ParallelPlan::solve(): Solution found by one or more threads in 0.010720 seconds
[ INFO] [1623080452.669057000]: SimpleSetup: Path simplification took 0.007539 seconds and changed from 3 to 2 states
[ INFO] [1623080452.672276100]: Disabling trajectory recording
`

- Terminal 2: ur5_bringup
`[ INFO] [1623080416.352382900]: Setting up connection: 192.168.0.11:30003
[ INFO] [1623080416.355665500]: Connection established for 192.168.0.11:30003
[ INFO] [1623080416.356115400]: Setting up connection: 192.168.0.11:30001
[ INFO] [1623080416.358892800]: Connection established for 192.168.0.11:30001
[ INFO] [1623080416.475540700]: Got VersionMessage:
[ INFO] [1623080416.475712900]: project name: URControl
[ INFO] [1623080416.475782600]: version: 3.6.0
[ INFO] [1623080416.475848600]: build date: 14-05-2018, 16:43:28
[ INFO] [1623080416.475899400]: Disconnecting from 192.168.0.11:30001
[ INFO] [1623080416.495510700]: ActionServer enabled
[ INFO] [1623080416.497285500]: Use standard trajectory follower
[ INFO] [1623080416.499393700]: Setting up connection: :50001
[ INFO] [1623080416.499708900]: Connection established for :50001
[ INFO] [1623080416.499849300]: Initializing ur_driver/URScript subscriber
[ INFO] [1623080416.503553000]: The ur_driver/URScript initialized
[ INFO] [1623080416.504963400]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1623080416.515754700]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1623080416.516036800]: Starting main loop
[ INFO] [1623080416.516348000]: Starting pipeline RTPacket
[ INFO] [1623080416.516884100]: Starting pipeline StatePacket
[ INFO] [1623080416.519282500]: Starting ActionServer
[ INFO] [1623080416.524401300]: Trajectory thread started
[ INFO] [1623080416.524862000]: Setting up connection: 192.168.0.11:30002
[ INFO] [1623080416.529830100]: Setting up connection: 192.168.0.11:30003
[ INFO] [1623080416.532862400]: Connection established for 192.168.0.11:30002
[ INFO] [1623080416.541596900]: Connection established for 192.168.0.11:30003
[ INFO] [1623080452.677787200]: Received new goal
[ INFO] [1623080452.677950900]: Trajectory received and accepted
[ INFO] [1623080452.678130500]: Translating trajectory
[ INFO] [1623080452.678212500]: Executing trajectory with 6 points and duration of 1.164s
[ INFO] [1623080452.678299500]: Attempting to start follower 0x55ab29491ae0
[ INFO] [1623080452.678372500]: Uploading trajectory program to robot
`

[... 3 lines cut ...]
[ INFO] [1623080439.117918900]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1623080439.130554200]: Forcing OpenGl version 0.
[ INFO] [1623080439.389506700]: Stereo is NOT SUPPORTED
[ INFO] [1623080439.389681100]: OpenGL device: llvmpipe (LLVM 8.0, 256 bits)
[ INFO] [1623080439.389802600]: OpenGl version: 3.1 (GLSL 1.4).
[... cut ...]
[ INFO] [1623080444.974143700]: Ready to take commands for planning group manipulator.
[ INFO] [1623080444.974282600]: Looking around: no
[ INFO] [1623080444.974468000]: Replanning: no
[ INFO] [1623080444.997639900]: Marker 'EE:goal_ee_link/10': mesh_use_embedded_materials is ignored.
`
So in rviz i can select the desired pose and click on "plan and execute" the planning part shows but not the execution, so i'm guessing the problem is similar as the terminal ouputs are pretty much the same. 

Thanks for any replies on the question!

angux94, 2021-06-07:
> Hello! I'm facing the same problem as the one in this post, but my robot is a UR5 series, I can't seem to find the "enable remote control" button or settings, but for the most part I have the same [...]

Sorry for the format in the terminals output, it's my first time commenting and not really sure how to format it :/

gavanderhoorn, 2021-06-07:
Please do not post new problems as comments on already closed issues.

Post new issues.

But, having written that: if you are not using a CB2 controller, there's nothing we can do here. You should be using [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues) in that case.