ros-industrial-attic/ur_modern_driver #325: Using robotiq 2F-140 gripper in an UR10e robot

Drikuy, 2019-07-03:
I'm new in ROS and yesterday I've started moving the UR10e robot using `ur_modern_driver`. I want to use an **2F-140 Robotiq Gripper**. I've read that in UR modern robots, the connection cable goes through the robot joint and links so you have no way to use a proper wire to connect with the gripper. So my question is how can I do the comunicatión with the Gripper? I guess is by using some kind of `io_interface` to manage tool I/O pins. This way you just communicate with the UR10e robot using TCP communication, and to control the gripper you just send the propper command through this TCP connection. Is there any example of how to do this communication?

miguelprada, 2019-07-03:
The e-Series robots do include a RS485 bus on the flange connector, but that doesn't mean you cannot route all the cables you want outside of the robot. How you should connect to your gripper depends a lot on the specifics of the gripper you have. Does your gripper only provide cabling to connect to the URe's flange or does it provide a more general purpose connector? I've worked with that gripper model on CB-Series robots, and I've always routed the wires outside the UR arm and to the controller. Newer models may offer different connectivity options, though.

Also, I'm not very familiar with the new RS485 interface, but there doesn't seem to be anything explicitly supporting it in either of the PRs adding support for the e-Series robots to this driver. It may very well be the case that you need specific URCap support to use that interface.

Since this is not an issue with the driver per se, I will close it, but feel free to continue the discussion here if you want.

Drikuy, 2019-07-03:
Ok thanks, I will try to control it by using URScripts commands.

closed 2019-07-03 by Drikuy