ros-industrial-attic/ur_modern_driver #330: can connect to UR10 but cannot control with moveit.

nag92, 2019-07-08:
I am trying to control the UR10 using moveit. I am have the vanilla ur_modern_driver and universal_robot packages. Both of which are on the kinetic-devel branches. 

When I run the following commands:

terminal 1: ```roslaunch ur_modern_driver ur10_ros_control.launch limited:=true robot_ip:=192.168.50.1 ``` 
terminal 2: ``` roslaunch ur10_moveit_config ur10_moveit_planninexecution.launch```

terminal 3: ```roslaunch ur10_moveit_config moveit_rviz.launch```

I am able to connect to the robot and echo the joints states but when I try to control the robot using the Rviz gui the arm wont move and is not in the same configuration as the phyical robot. 
My terminal oututs are as follows: 

Terminal 1: 
```
... logging to /home/nathanielgoldfarb/.ros/log/21c926f6-a19d-11e9-9aa3-9c2a70161d03/roslaunch-RISE-Dell4D-13769.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://127.0.0.1:34495/

SUMMARY
========

PARAMETERS
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 125
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_based_pos_traj_controller/action_monitor_rate: 10
 * /pos_based_pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/elbow_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/constraints/goal_time: 0.6
 * /pos_based_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_based_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_based_pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.1
 * /pos_based_pos_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_based_pos_traj_controller/state_publish_rate: 125
 * /pos_based_pos_traj_controller/stop_trajectory_duration: 0.5
 * /pos_based_pos_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /ur_hardware_interface/base_frame: base
 * /ur_hardware_interface/max_payload: 3.0
 * /ur_hardware_interface/max_velocity: 10.0
 * /ur_hardware_interface/min_payload: 0.0
 * /ur_hardware_interface/prefix: 
 * /ur_hardware_interface/reverse_ip_address: 
 * /ur_hardware_interface/reverse_port: 50001
 * /ur_hardware_interface/robot_ip_address: 192.168.50.1
 * /ur_hardware_interface/servoj_gain: 750.0
 * /ur_hardware_interface/shutdown_on_disconnect: True
 * /ur_hardware_interface/tool_frame: tool0_controller
 * /ur_hardware_interface/use_ros_control: True
 * /vel_based_pos_traj_controller/action_monitor_rate: 10
 * /vel_based_pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /vel_based_pos_traj_controller/constraints/elbow_joint/trajectory: 0.1
[... 343 lines cut ...]
SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /rviz_RISE_Dell4D_17938_6059954651304162092/manipulator/kinematics_solver: kdl_kinematics_pl...
 * /rviz_RISE_Dell4D_17938_6059954651304162092/manipulator/kinematics_solver_attempts: 3
 * /rviz_RISE_Dell4D_17938_6059954651304162092/manipulator/kinematics_solver_search_resolution: 0.005
 * /rviz_RISE_Dell4D_17938_6059954651304162092/manipulator/kinematics_solver_timeout: 0.005

NODES
  /
    rviz_RISE_Dell4D_17938_6059954651304162092 (rviz/rviz)

ROS_MASTER_URI=http://127.0.0.1:11311

process[rviz_RISE_Dell4D_17938_6059954651304162092-1]: started with pid [17955]
[ INFO] [1562603785.091453985]: rviz version 1.12.17
[ INFO] [1562603785.091494860]: compiled against Qt version 5.5.1
[ INFO] [1562603785.091504523]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1562603785.950358276]: Stereo is NOT SUPPORTED
[ INFO] [1562603785.950453704]: OpenGl version: 4.5 (GLSL 4.5).

```

Drikuy, 2019-07-11:
This closed issue may help you #317. I had the same problem and it was because I didn't switch on properly the "Remote Control" mode on the teaching pendant of robot.

gavanderhoorn, 2019-11-23:
It's somewhat unclear from the logs shown, but if this is a CB3 or e-series controller, please use `ur_robot_driver` instead of `ur_modern_driver`.

closed 2019-11-23 by gavanderhoorn