ros-industrial-attic/ur_modern_driver #331: Action client not connected in ROS Melodic

vkguda355, 2019-07-12:
Hi,

I have the melodic version of the ur5 package 

I can properly connect to the robot typing:

```
roslaunch ur_moder_drivers ur5_bringup.launch robot_ip:=IP_OF_THE_ROBOT 
```

however, when I try to run the moveit planner typing:
```
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch 
```

I get the error messange:

```
[ERROR] [....]: MoveitSimpleControllerManager: Action client not connected: /follow_joint_trajectory
```

How can I fix this issue?
This issue, by the way, it was working few hours back without any problem.

I followed the tutorial : http://wiki.ros.org/universal_robot/Tutorials/Getting%20Started%20with%20a%20Universal%20Robot%20and%20ROS-Industrial 

thanks for helping

gavanderhoorn, 2019-07-12:
I've also transferred this to `ur_modern_driver` for now, as because of the missing action topics it would appear to be a driver issue, not a MoveIt configuration one.

vkguda355, 2019-07-12:
the problem seems to have been fixed, i had to restart my robot. I ran the code ur_modern_driver ur5_bringup.launch, to see the ros topics listed. 

`/joint_states
/joint_temperature
/rosout
/rosout_agg
/tf
/tf_static
/tool_velocity
/ur5_manipulator_controller/follow_joint_trajectory/cancel
/ur5_manipulator_controller/follow_joint_trajectory/feedback
/ur5_manipulator_controller/follow_joint_trajectory/goal
/ur5_manipulator_controller/follow_joint_trajectory/result
/ur5_manipulator_controller/follow_joint_trajectory/status
/ur_driver/URScript
/ur_driver/io_states
/ur_driver/masterboard_state
/ur_driver/robot_mode_state
/ur_driver/robot_status
/wrench
`

still any idea why i was having this issue yesterday and got resolved, today ?

gavanderhoorn, 2019-07-12:
> still any idea why i was having this issue yesterday and got resolved, today ?

No. This could be an interplay between the robot controller and the driver, but without much more information and on-site debugging, I doubt we'll be able to figure out what was going on.

gavanderhoorn, 2019-07-12:
Closing as it seems to be fixed.

gavanderhoorn, 2019-07-12:
If you haven't done so already, I would suggest to make sure you run the most up-to-date code.

closed 2019-07-12 by gavanderhoorn