ros-industrial-attic/ur_modern_driver #332: UR e-series monitoring

Drikuy, 2019-07-12:
Hi everyone, I'm not sure if this is the place to ask this question but I think it is. Using `ur_modern_driver` can you read the current or the torque of all the joints?. I know that ther are UR Script functions in order to do that, but my question is if there is already a topic that publish it when you launch `ur_modern_driver ur10e_bringup.launch robot_ip:=[ROBOT_IP]` or do I have to do it by myself writing my own ROS node?

gavanderhoorn, 2019-07-12:
We currently publishes joint motor currents in the `effort` field of the [JointState](http://docs.ros.org/api/sensor_msgs/html/msg/JointState.html) messages ([here](https://github.com/ros-industrial/ur_modern_driver/blob/fd2e38af745b9d471a4f9532be5f39e5d95fe405/src/ros/rt_publisher.cpp#L29)).

This data comes from the controller over the real-time interface, which only includes the `I_actual` field. There is no `M_actual` field corresponding to `M_target`.

There is no analogue to the `get_joint_torques()` urscript function in that datastream afaict, so we cannot publish it.

gavanderhoorn, 2019-07-12:
Closing as it's not an actionable issue at the moment.

closed 2019-07-12 by gavanderhoorn

Drikuy, 2019-07-12:
Thank you for your explanation!