ros-industrial-attic/ur_modern_driver #335: Launching the driver causes the arm to go to HOME position (all joints equal to 0)

joaocabogon, 2019-07-16:
Hello,

When the driver is launched the arm, goes to the HOME joints configuration (all zero) in a dangerous way.

How can I launch the driver and keep the arm in the current configuration? In other words, I want to launch the driver without moving the arm.

Thanks

gavanderhoorn, 2019-07-16:
Please clarify which branch you have built: `master` or `kinetic-devel`.

joaocabogon, 2019-07-16:
I am actually using this branch: https://github.com/plusone-robotics/ur_modern_driver/tree/ur_moderner_driver

I am using: `roslaunch ur_modern_driver ur5e_ros_control.launch robot_ip:=...`

Using a UR5e arm with Polyscope 5.3.1

gavanderhoorn, 2019-07-16:
> I am actually using this branch: https://github.com/plusone-robotics/ur_modern_driver/tree/ur_moderner_driver

Then I'm rather confused as to why you are posting an issue here.

We do not support forks of this repository.

> I am using: `roslaunch ur_modern_driver ur5e_ros_control.launch robot_ip:=...`

Ah. `ros_control` mode.

It's possible that fork does not include #213, which *could* cause this.

But I have no idea, as I don't use that fork.

gavanderhoorn, 2019-07-16:
As you are using a fork and not the package in this repository, I'm going to close the issue.

Please go to the maintainers of the fork for support.

Feel free to keep commenting on this issue of course.

closed 2019-07-16 by gavanderhoorn

joaocabogon, 2019-07-17:
Commenting just to thank you and to say that #213 solved the problem.