ros-industrial-attic/ur_modern_driver #336: "Low Bandwidth Trajectory Follower" cannot connect to robot

xarthurx, 2019-07-17:
I experienced a "catch-up" issue:

when using the UR5 to do Cartesian planning, the robots sometimes execute behind the planned time and will miss some of the necessary points in order to "catch-up".

So I tried to use the `Low Bandwidth Trajectory Follower` from the readme.

However, when sending several paths (each contains a series of frames to be planned by moveit and send to the robot) together, the robot only executes 1-2 of them and then stop, with information shown below:

UR panel: (A lot of them when using `lowbandwidth`, none when not)
[image]

Ros: (Guess this is the reason why the robot stop)
[image]

Any suggestions to fix?

gavanderhoorn, 2019-08-27:
> However, when sending several paths (each contains a series of frames to be planned by moveit and send to the robot) together

it's unclear to me what you're doing here exactly.

How do you "send[ing] several paths"? And to what/where/who?

Trajectory replacement is not supported.

xarthurx, 2019-08-27:
Sorry for the confusion.

I'm sending a series of trajectory paths for the robot to follow the Cartesian trajectory, one after another.

Not replacement, but just one after another.
In the normal mode, these operations are fine, but when using `low_bandwidth`, only the first one will be followed and then the robot will stop.

gavanderhoorn, 2019-08-27:
> Not replacement, but just one after another.

Ok. Thanks for the clarification.

> In the normal mode, these operations are fine, but when using `low_bandwidth`, only the first one will be followed and then the robot will stop.

Does the action server indicate that trajectory execution was complete? Or does it hang, waiting for the motion to complete?

xarthurx, 2019-08-28:
It doesn't hang.
I can continue to do other planning jobs /actions.

gavanderhoorn, 2019-08-28:
I believe this is going to be difficult to diagnose unless we have a way to reproduce it.

Could you provide a bit more information on how you have things setup, which versions of components, where are you submitting the trajectories (to MoveIt's execute trajectory action? To the driver directly? Somewhere else?) and other pertinent details.

gavanderhoorn, 2019-09-04:
If this is still a problem and you'd like to diagnose this it would be good if you could provide the information requested in my previous comment.

xarthurx, 2019-09-04:
I'm not using the low_bandwidth currently but can provide more info, if you
can keep this issue open.
Just a little busy recently.

gonzalocasas, 2019-09-16:
> No, it can be run at a lower rate (see @ThomasTimm's report)

@gavanderhoorn and what would that entail, roughly? I assume there's no parameter `ur_control_rate_in_hz` somewhere lying around, so, would it mean tweaking the code (possibly in a multitude of places) and testing if it still behaves?

gavanderhoorn, 2019-09-16:
You would have to change these parameters:

https://github.com/ros-industrial/ur_modern_driver/blob/fd2e38af745b9d471a4f9532be5f39e5d95fe405/launch/ur10_bringup.launch#L20-L25

and make the [configuration for `ros_control` match](https://github.com/ros-industrial/ur_modern_driver/blob/fd2e38af745b9d471a4f9532be5f39e5d95fe405/config/ur10_controllers.yaml).

Note: that's all for the non-lbtf implementation though. The LBTF does not use those iirc.

It's non-trivial to tweak those values to something other than what they are now though, and I wouldn't be able to provide assistance (not because I wouldn't want to, but because it's been too long since I've done anything with those values).

gonzalocasas, 2019-09-16:
Understood! Thanks!

omar-enein-nist, 2022-01-14:
> It's unfortunate that the problems are now no longer there.
> 
> Performance in VMs has always been difficult, and VMs on Windows even more so.
> 
> The network infrastructure in VMs is sometimes not up to the task.
> 
> Just to learn something from this: which VM software were you using? VirtualBox, VMWare something else?

Hello,

I recognize that this thread has been inactive for some time and that ur_modern_driver has since been deprecated, however I have encountered a similar error in using the Low Bandwidth Trajectory Follower, and believe that I have uncovered more information regarding this specific problem. I am currently using the latest kinetic-devel branch of ur_modern_driver with a UR5 CB3 robot through ROS Noetic and using WSL1 (I also had to communicate with the UR5 over Wi-Fi). Similar to the original poster, I also experienced similar "catch up" issues with Cartesian planning using the default trajectory follower and subsequently attempted to enable the Low Bandwidth Trajectory Follower (LBTF). The first trajectory attempted by my program would execute correctly with the robot, however subsequent trajectories would fail with the same error messages shown in the ROS console output of the original poster. 

Adding additional error output inside the if statement starting at line 77 of server.cpp, I discovered that the error was caused by the client socket file descriptor (client_.getSocketFD()) still being valid when the accept() method was called to initialize a new socket. Generally, it seemed that each time a trajectory is executed, a new client TCP socket would be created, the trajectory would then be sent to the controller over the socket, and then the socket would be closed (this in trajectoryThread() starting at line 280 of action_server.cpp). I also discovered that when the first trajectory was executed, and the stop() method was called inside trajectoryThread() (line 356 of action_server.cpp), the disconnectClient() method was not being called because the client socket state had already been set to "Disconnected". 

I determined the problem was that, in the read() method starting at line 143 of tcp_socket.cpp, there is an if statement checking whether the call to recv() returns 0, which indicates the client socket had an orderly disconnect (https://man7.org/linux/man-pages/man2/recv.2.html). The client socket state is then updated to "Disconnected", but the file descriptor is not also reset to an invalid value. Therefore, to fix the problem on my end, I added an additional line below 154 of tcp_socket.cpp to set the file descriptor to -1 (i.e., "socket_fd_ = -1"). I am not sure whether or not there is a better general fix, but I hope this helps!