ros-industrial-attic/ur_modern_driver #337: UR5e stopping for half a second when executing a trajectory

joaocabogon, 2019-07-17:
Hello,

So I am having this problem of the arm (UR5e) stopping for something like half a second, doing a cracking sound and then proceeds on the trajectory it was executing.

I am using a UR5e with PolyScope 5.3.1, ROS Kinetic, Ubuntu 16.

When executing trajectories the arm SOMETIMES does these cracking sounds (seems like they are coming from the brakes). I thought it had something to do with the Payload and Center of Gravity, but when I adjusted these values, it improved but keeps making these sounds. In the worst case scenario, I have a Protective Stop where the arm really stops executing the trajectory:

`C157A1: Collision detected by joint: (shoulder). Check payload, center of gravity and acceleration settings.`

I am using ros_control and the vel_based_pos_traj_controller with the values of the 
`ur5_controllers.yaml` on this repo. 

Besides, I am using the /ur_driver/URscript topic to send commands like set_tool_voltage or set_payload and somehow I think this might be related with the problem.

Any suggestions on how to solve this?

gavanderhoorn, 2019-07-17:
In #335 you were using a fork. Is that still the case?

joaocabogon, 2019-07-17:
Yes, do you think it may have to do with it? 

I am using the fork because I was getting the "Sub-package of type 0 was not parsed completely" with my UR5e arm.

gavanderhoorn, 2019-07-17:
Please stop posting issues about forks on this tracker.

We cannot help you with problems you encounter with a fork, as we don't know what the fork has changed.

Please seek support from the maintainers/authors of the fork.

gavanderhoorn, 2019-07-17:
#216 is a PR that adds preliminary support for some e-series robots.

The new driver currently in development by FZI & UR will have full support for e-series robots.

closed 2019-07-17 by gavanderhoorn

joaocabogon, 2019-07-17:
> Please stop posting issues about forks on this tracker.
> 
> We cannot help you with problems you encounter with a fork, as we don't know what the fork has changed.

Ok, it was just because (as I mentioned in the other issue) the fork as no Issues tab to report these things.

> The new driver currently in development by FZI & UR will have full support for e-series robots.

Any idea when this will be ready?

gavanderhoorn, 2019-07-17:
> > The new driver currently in development by FZI & UR will have full support for e-series robots.
> 
> Any idea when this will be ready?

Please see [ROS-Industrial Conference 2018: Official ROS support: making the most out of the new UR e-Series](https://www.youtube.com/watch?v=gjrnOdymUUE&list=PLXUpEXjGC63z7Zhe0gyk2tCkZvoEsADpg&index=7&t=0s).

gavanderhoorn, 2019-07-17:
> > Please stop posting issues about forks on this tracker.
> > We cannot help you with problems you encounter with a fork, as we don't know what the fork has changed.
> 
> Ok, it was just because (as I mentioned on the other issue) the fork as no Issues tab to report these things.

That is unfortunate, but not something we can help you with.