ros-industrial-attic/ur_modern_driver #340: "FreeDriver" of UR5

ghx-0228, 2019-08-04:
Can I use this driver to extract the "freedrive" signal (off/on) from real UR5?

gavanderhoorn, 2019-08-27:
I can't verify whether we're publishing it already, but the RT state data contains a field `robotMode` that encodes what mode the robot is in. `BACKDRIVE` or `FREEDRIVE` would indicate whether the freedrive button is being pressed (or actually: whether the robot is actually in that mode. It may refuse to enter freedrive mode).

gavanderhoorn, 2019-08-27:
Closing as it's not an issue with the package in this repository (right now).

If you can't make this work, please log as an enhancement -- and submit a PR if you'd like this supported. Otherwise it may take a while before this is implemented.

closed 2019-08-27 by gavanderhoorn