ros-industrial-attic/ur_modern_driver #344: Using URCaps in TrajectoryFollowers

Brudasoe, 2019-08-29:
Hi,
I've been thinking about modifying TrajectoryFollowers to include a way to control the Robotiq grippers, but it seems like functions that are provided by URCaps (like rq_move) are unavailable when sending scripts over this interface. Is that intended design? Or is there a way to include the libraries that I need for additional equipment? 
I know that when saving regular scripts, an additional file is created, containing all the additional functions used. Is that mechanism available for saving the sent scripts?
The main reason I want to do that is so that I can connect the gripper directly to robot, removing unnecessary cables to PC (maybe there is another way to do this, but this came to me first).

gavanderhoorn, 2019-09-04:
I don't believe this was a concern/requirement when the trajectory followers were designed, so it may well be that this is difficult to achieve.

> but it seems like functions that are provided by URCaps (like rq_move) are unavailable when sending scripts over this interface.

"over this interface" is vague. Are you referring to the `urscript` topic? In the end that sends your URscript directly to the controller (there is no magic wrapping or session management), so it may well be that this doesn't support calling functions contributed by urcaps.

gavanderhoorn, 2019-09-04:
As this does not appear to be an actionable item for this repository, I'm going to close it.

If it turns out we can turn this into something actionable, we could re-open.

In the meantime: please feel free to keep commenting on the issue.

closed 2019-09-04 by gavanderhoorn

Brudasoe, 2019-09-04:
Thanks for the information!
By the interface I meant the way TrajectoryFollower script is sent. So I guess for now that is unavailable, except maybe if I added the whole code gripper uses. Is there a possibility that in the future an include mechanism could be added to URScript language?

gavanderhoorn, 2019-09-04:
>  Is there a possibility that in the future an include mechanism could be added to URScript language?

I really wouldn't know. I'm not a UR employee, so you'd have to ask UR about this.